OrcaSlicer/OrcaSlicer #1863: Presets lost when closing OrcaSlicer

alexcagliari87, 2023-08-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. Open OrcaSLicer 1.6.4 beta 3 from .exe file
2. Select a printer, a filament and a profile
3. In process tab make some edits in current profile
4. Click on save button and assign a name
5. Close OrcaSlicer
6. Reopen

**Expected behavior**
I need to find the previously saved profile

**Actual behavior**
Can't find previously saved profile

**Printer model**
FLSun V400

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 10.0.19045 Build 19045

mysugarape, 2023-08-21:
Exactly the same issue

Desktop:
OS: macOS Big Sur
Version: 11.6 (20G165)

Glenn1963, 2023-08-21:
I have a similar issue and am seeing that the saved profile gets deleted as soon as I start Orcaslicer. Looking at the log, it is failing on my Post processing script entry, which was valid and worked correctly before I closed Orca.

```
2023-08-21 15:31:15.650731[Thread 0x000004a8]:Slic3r::ConfigBase::load_from_json: parse C:\Users\viewt\AppData\Roaming\OrcaSlicer\user\default\process\0.48mm Fast V400 Glenn.json got a generic exception, reason = Invalid value provided for parameter post_process: ""L:\\Program Files\\Python311\\python.exe" "Q:\\Development\\Pythons\\SlicerScripts\\m73.py""
2023-08-21 15:31:15.650731[Thread 0x000004a8]:parse config C:\Users\viewt\AppData\Roaming\OrcaSlicer\user\default\process\0.48mm Fast V400 Glenn.json failed
2023-08-21 15:31:15.654730[Thread 0x000004a8]:update_compatibility for all presets enter
```

Glenn1963, 2023-08-24:
Just following up on this, Orca seems to hate a Post processing entry that has spaces (and is therefore quote delimited) in the path. It saves to the json file but fails to load the next time and deletes the profile. I suspect the json is written with invalid escaping for all the many quotation marks and slashes it tries to stick in there.

In summary, Orca deletes a profile when it tries to read it and fails.

alexcagliari87, 2023-08-24:
I confirm the point about spaces.
I added this post-processor but after I saved it correctly and closed the software, when I reopen it, I no longer find the setting.

[image]

@SoftFever please can you fix it? it is getting really tedious to lose settings when closing Orcaslicer..
This behavior happens even with the newly updated version.
Thank you

ZeroDayExploits, 2023-09-05:
I did some debugging and have a fix.

Here are my notes for the problem
(I use of single quots to denote string to avoid confusion: 'some string')

- Config that contains:
 ```
"post_process": [
        "\"Hi   d \""
    ]
```

- Config is read by ConfigBase::load_from_json
- entry ist String array
- Some code that converts the String array to a single String
```
for (auto iter = it.value().begin(); iter != it.value().end(); iter++) {
        if (iter.value().is_string()) {
            if (!first) {
                if (use_comma) // use_comma == False
                    value_str += ",";
                else
                    value_str += ";";
            }
            else
                first = false;

            if (use_comma)
                value_str += iter.value();
            else {
                value_str += "\"";
                value_str += escape_string_cstyle(iter.value());  
                value_str += "\"";
            }
        }
```
- after that value_str = '""Hi   d ""' // outer quotes from array entry inner quotes from actual string
- this->set_deserialize(opt_key, value_str, substitution_context);
  in ConfigBase::set_deserialize_raw:
  the ConfigOptionDef for post_process has the type coStrings
 ->use unescape_strings_cstyle to parse value_str = '""Hi   d ""'
- Code snippet form unescape_strings_cstyle 
```
c = str[i];
        if (c == '"') {
            // Complex case, string is enclosed in quotes.
            for (++ i; i < str.size(); ++ i) {
                c = str[i];
                if (c == '"') {
                    // End of string.
                    break;
                }
...
```

- here is the error unescape_strings_cstyle expects unescaped strings in the quotes ie '"\"string \""',
	but escape_string_cstyle does not replace " with \"
				
- adding the following to escape_string_cstyle will fix the issue, but may cause different problems.
```
else if (c == '\"') {
	(*outptr ++) = '\\';
	(*outptr ++) = '\"';
}
```

ZeroDayExploits, 2023-09-05:
PR: #2014

closed 2023-10-01 by SoftFever