OrcaSlicer/OrcaSlicer #1869: Adding or removing a pause causes model to visually disappear

Orbitally, 2023-08-21:
**Describe the bug**
After slicing a model when moving the layer slider to a specific later and clicking the button to add a pause the model(s) being sliced to disappear off the build place until re-slicing the plate. Even though the model disappears off the plate visually the plate preview thumbnail in the top left still shows the model even though it's no longer shown on the build plate. 

**3mf File for This Bug**
Happens with any model including any of the included primitive models.

**To Reproduce**
1. Add any model to the build plate
2. Slice the model
3. Pick any layer and add a pause to that layer
4. The model with disappear off the build plate
5. Click the `Slice place` button again to make it reappear
6. Repeat for removing a pause

**Expected behavior**
The model should stay visible on the plate and not wander off to explore the void.

**Screenshots**
Just after slicing:
[image]

Just added a pause:
[image]

Changed nothing and just re-clicked the slice button again:
[image]

**Printer model**
Ender 3 v2 (Don't really think it's that applicable in this case)

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 22H2

Orbitally, 2023-08-22:
Just a quick update that it seems like changing any parameter causes this issue to pop up and this is on v1.6.4 Beta 3

Linnie73, 2023-09-01:
V1.6.6: model disappears from view in the Preview tab on every slice action. Regardless of parameters changed, Happens with any model.

Orbitally, 2023-09-01:
Also still an issue for me with 1.6.6 as well

closed 2023-12-09 by github-actions[bot]