OrcaSlicer/OrcaSlicer #1875: AMS filament dropdown list broken / empty

nirurin, 2023-08-22:
**Describe the bug**
Go to device page, look at a filament in the AMS, and click the dropdown in order to select what filament is in that slot. Instead of a list of default filaments, the dropdown doesn't open, and is blank. 

**Expected behavior**
It should show the default list of filaments from bambu and generics. 

**Screenshots**
[image]

**Printer model**
P1S

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 1122H2

SurgeHawk, 2023-08-22:
The fix for this should be in 1.6.4, released today, 3 hours ago. But this is what fixed it for me #1712

SoftFever, 2023-08-24:
It should be fixed in 1.6.6.