OrcaSlicer/OrcaSlicer #1877: [1.6.4] Not connecting to server

Ohmarinus, 2023-08-22:
**Describe the bug**
The Beta version and Bambu Studio are both connecting to my machine fine but the latest version of Orca doesn't want to 'connect to server'. I tried both the portable and the installer version for Windows but no connection can be made. I chose to allow the software to connect to private networks, just like I always do.

I am using the latest version of Bambu Studio and have installed the network plugins correctly and it's all working normally.

**To Reproduce**
Start the software, go to the Device pane and there it is:
[image]

It sees the machine, but it can't connect to the server. Despite the beta version and BS being able to connect perfectly fine.

**Expected behavior**
To be able to send a print

**Screenshots**
[image]

**Printer model**
Bambu Lab X1C

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 22H2

FlyingLemming, 2023-08-22:
Same here, running on macOS 13.5

STC148, 2023-08-22:
I have the same problem, win11

auafuma, 2023-08-22:
I had same issue, logged out of my bambu account and back in again, then I was able to connect fine.

Ohmarinus, 2023-08-22:
> Sorry, does not seem to be solved at all. Logging out and in again works fine for the moment. However, after restarting OrcaSlicer, the issue is there again. I need to repeat the procedure each and every time I launch the app.
> 
> macOS 13.5, Apple Silicon

Thanks for letting me know, I just found this out myself as well, I reopened the issue and updates the description.

CoryCharlton, 2023-08-22:
Same issue here :(

shyblower, 2023-08-22:
Logging in and out does not work for me on Linux. Maybe because Orca crashes on logout.

FlyingLemming, 2023-08-22:
On macOS, I initially thought that OrcaSlicer has crashed upon logout. However, it just took one or two minutes, then it succeeded. All further logout attempts took only a second.

capsel22, 2023-08-22:
can confirm, logging out of Orca, closing down, opening, logging  back in worked.

FlyingLemming, 2023-08-22:
Then close Orca once more and restart – it will most probably not work anymore.

Ohmarinus, 2023-08-22:
> can confirm, logging out of Orca, closing down, opening, logging back in worked.

As stated, the settings disappear after closing and reopening Orca again after fixing it. It doesn't seem to get stored correctly.

capsel22, 2023-08-22:
Yes correct, issue returns after re-opening Orca.

psiberfunk, 2023-08-22:
Same problem here @SoftFever .  Seems that this has been reproduced by lots of people at this point.

Macmagoo, 2023-08-22:
Same problem here. Camera feed works but no server connection.

psiberfunk, 2023-08-22:
possible candidates commits:

- https://github.com/SoftFever/OrcaSlicer/commit/6e1bdaf9d43f0aa6be7e881d03e9b3154748735e
- https://github.com/SoftFever/OrcaSlicer/commit/9e73dfeb5d32ad3b1020137f3ee74d826c3a9470

psiberfunk, 2023-08-22:
last valid working windows nightly build appears to be https://github.com/SoftFever/OrcaSlicer/actions/runs/5915113014 folks

msteele999, 2023-08-22:
same issue here - Windows 11 Home.

Only seems to affect my X1CC and not my Voron

SoftFever, 2023-08-22:
Thanks guys.
I'm aware of this issue, will investigate and release hotfix soon

SoftFever, 2023-08-22:
> last valid working windows nightly build appears to be https://github.com/SoftFever/OrcaSlicer/actions/runs/5915113014 folks. All main branch builds (on github.. i don't have a windows build environment) after this one fail for this bug. I suspect this commit is the problem @SoftFever : [6e1bdaf](https://github.com/SoftFever/OrcaSlicer/commit/6e1bdaf9d43f0aa6be7e881d03e9b3154748735e)

Thanks.
I have pinpointed the issue, but I can only work on it tonight after work. Should be able to release a hotfix today or tomorrow

FlyingLemming, 2023-08-23:
No hurries. 🤗
Thank you for your efforts and this magnificent piece of software!

swdw973, 2023-08-23:
Glad they reopened this. Using a work around does not constitute solving the problem.  

Even more glad they found the issue so they can fix it. Great job!!!

SoftFever, 2023-08-23:
Hey guys,

Can someone help me verifying if this issue is fix with this build(it should be ready in 25 minutes)?
https://github.com/SoftFever/OrcaSlicer/actions/runs/5952301579

Naninani, 2023-08-23:
@SoftFever the fix is working for me on Apple silicon.

Ohmarinus, 2023-08-23:
> Hey guys,
> 
> Can someone help me verifying if this issue is fix with this build(it should be ready in 25 minutes)? https://github.com/SoftFever/OrcaSlicer/actions/runs/5952301579

Connections seems to be made for me! Uploading and printing something now :)

No need to relog within Orca anymore, so for me, I can confirm at least the Win10 portable version works.

FlyingLemming, 2023-08-23:
Same here. Started the macOS/ARM app and the printer has been detected right away.

SoftFever, 2023-08-23:
I will bump the version then.
Will publish a new hotfix tomorrow.

msteele999, 2023-08-23:
I can test Windows version in 30 minutes

sent from my mobile device.

capsel22, 2023-08-23:
> I will bump the version then. Will publish a new hotfix tomorrow.

thanks for the quick fix!

SurgeHawk, 2023-08-23:
Windows seem to be working now

msteele999, 2023-08-23:
Confirmed working for me as well

Ohmarinus, 2023-08-23:
Since this issue is resolved, I will update the issue and close this issue. Thanks for the quick fix @SoftFever

closed 2023-08-23 by Ohmarinus