OrcaSlicer/OrcaSlicer #1878: "Connecting to Server" Error when trying to send print - 1.6.4 release

TheSephiCloud, 2023-08-22:
"Connecting to Server" Error when trying to send print

Slice anything and then try to "Print Plate" 
It won't let you send even though it sees the printer

Tried with the Windows Installer aswell as the portable Version
Stealth Mode is off and turning it off and on did not fix it either

Bug did not exist in 1.6.4.b2

P1P

Desktop Win10

ThoughtfulDev, 2023-08-22:
Having the same issue. Didnt exist in b3

Naninani, 2023-08-22:
Go to the Device tab and you will see it. I get the same as you on the Preview tab - Its the same issue.
[image]

ThoughtfulDev, 2023-08-22:
After logging out and then loggin back in it works...until you close orcaslicer.
I get both errors (on device tab AND on sent).

NickUfer, 2023-08-22:
+1 same on windows

FlyingLemming, 2023-08-22:
This is the same issue as [#1877](https://github.com/SoftFever/OrcaSlicer/issues/1877).

Aron3DP, 2023-08-22:
I also cannot connect to any of my Bambu printers after updating. Reverted back to Beta 3 and now I can. BambuStudio also connects fine as well.

SoftFever, 2023-08-22:
Thanks guys.
I'm aware of this issue, will investigate and release hotfix soon

closed 2023-08-24 by TheSephiCloud