OrcaSlicer/OrcaSlicer #1879: Camera view in LAN mode

Prutsium, 2023-08-22:
With the release of 1.6.4 there is still no camera view in LAN mode possible where BS can do this already.
[image]

Is this on the roadmap?

SurgeHawk, 2023-08-22:
What printer are you using? I have a P1S in LAN mode and the cameras works fine in Orca.

Prutsium, 2023-08-22:
X1C with AMS

SoftFever, 2023-08-24:
can you try 1.6.6?

Prutsium, 2023-08-24:
Will download now

Prutsium, 2023-08-24:
Not sure what happened here but printer is online and connected (i see temp status and so on) 
When starting cam 2 errors ...
1: Not supported in LAN
2: Printer not conected :)
[image]

fire1ce, 2023-08-24:
same here. lan-only mode. OrcaSlicer v1.6.6.
RTSP steam works fine in other places from the printer.

SoftFever, 2023-08-25:
The access code will change everytime we change LAN or WAN mode.
What happens when you input the new acess code?

I tested on my P1P, it works fine.

Qonstrukt, 2023-08-25:
I have the same issue. I have the RTSP stream working in Home Assistant, but OrcaSlicer can't seem to open it. Trying to open it in another video client at the same time, like VLC, works just fine, with the same access code.

MotoFreak, 2023-08-25:
> The access code will change everytime we change LAN or WAN mode. What happens when you input the new acess code?
> 
> I tested on my P1P, it works fine.

In 1.6.6 I can get a video feed from my P1S  but not my X1C and both are in lan only mode.

STC148, 2023-08-25:
Here the same, no video connection with my X1C in LAN mode with 1.6.6

WZ9V, 2023-08-26:
Another one here, no camera feed from X1C LAN mode on Orca Slicer 1.6.6.  Running Linux Mint

NairoDorian, 2023-08-27:
Camera Live feed does not work in 1.6.6 with X1 as well, I'm still in 1.5 firmware with the printer though, does this feature need the printer to be updated as well ?

SoftFever, 2023-08-27:
It's tough to pinpoint the exact changes BL made in the slicer, closed-source plugin, or the black box printer firmware that might affect the camera feed.
Rather than chasing that down, I've updated the BL-related parts based on BS 1.7.4. 
This update also includes their lidar calibration changes.
Could you help verify if everything's working as expected?

You can download the nightly build here:
https://github.com/SoftFever/OrcaSlicer/actions/runs/5989893116

NairoDorian, 2023-08-27:
@SoftFever Tried your latest build you linked, now I get a warning telling me to upgrade the printer
[image]
I'll finish my print and will update the firmware and try that again, will update here once I have results

MotoFreak, 2023-08-27:
@SoftFever  just downloaded and tested the nightly build portable, P1S and X1C video feed works  now, thank you

STC148, 2023-08-27:
confirm! works on X1C now. Thank you

Prutsium, 2023-08-27:
Unfortunately cant check my printers are 12000km away as i am on business
trip in Sydney. If i find time will try to a vpn to home to test.

NairoDorian, 2023-08-27:
I got an error now I updated and rebooted the printer and activated the Lan Live Video View in the X1 menu
[image]

-10 | Certificate error | Check if you are using the official version of Bambu Studio

Qonstrukt, 2023-08-31:
> It's tough to pinpoint the exact changes BL made in the slicer, closed-source plugin, or the black box printer firmware that might affect the camera feed. Rather than chasing that down, I've updated the BL-related parts based on BS 1.7.4. This update also includes their lidar calibration changes. Could you help verify if everything's working as expected?
> 
> You can download the nightly build here: [SoftFever/OrcaSlicer/actions/runs/5989893116](https://github.com/SoftFever/OrcaSlicer/actions/runs/5989893116)

This works perfectly for my on my X1C in LAN mode, thanks 😄

bani6809, 2023-09-05:
> It's tough to pinpoint the exact changes BL made in the slicer, closed-source plugin, or the black box printer firmware that might affect the camera feed. Rather than chasing that down, I've updated [...]

this seems to have fixed it for me on my x1c

simonbuehler, 2023-09-08:
Video stream works for me now with the nightly on the p1p with new firmware, thanks!

WZ9V, 2023-11-20:
Camera mode starts but locks up after a few seconds.
Orca Slicer 1.8.0 on Linux Mint 21.2 Cinnamon

closed 2024-02-29 by github-actions[bot]