OrcaSlicer/OrcaSlicer #1880: Bridge not connected to perimeters and overhangs use bridge speed instead of overhang speed

celtare21, 2023-08-22:
**Describe the bug**
While slicing a bridge test, the gcode preview clearly shows that the bridge has a gap on both sides of the bridge, not being connected to the overhanging perimeters.
At the same time, the overhanging perimeters follow the "External bridge speed" instead of following the "Overhang speed"

**3mf File for This Bug**
(Couldn't upload it here so I got it on WeTransfer)
https://we.tl/t-5Resm3QWAT

**To Reproduce**
Just slice the model.
To see the incorrect speed, just change "External bridge speed" speed

**Expected behavior**
The bridge to be connected to the overhanging perimeters.
The speed of the perimeters to be the one from "Overhang speed", not the bridge speed

**Screenshots**
[image]
[image]

**Printer model**
Zero G Mercury One

**Desktop (please complete the following information):**
 - OS: Windows
 - Version Official 1.6.4 release

gbyleveldt, 2023-08-22:
This is something we've often complained about. I've been told that if an overhang is more than 90% of the area, it'll use bridge speed instead of overhang speed

celtare21, 2023-08-22:
> This is something we've often complained about. I've been told that if an overhang is more than 90% of the area, it'll use bridge speed instead of overhang speed

Imo that's a very bad choice. There's a reason we have bridge speed and overhang speed, no reason to mix them up, specially as it used the overhang speed for like 2mms then goes to the bridge speed. Such drastic and fast speed changes are very undesirable

gbyleveldt, 2023-08-22:
I hear you man - I agree. Like I said, it's been complained about lots of times. Personally, I run my bridges pretty fast, but on long overhang runs I want it slow. Like you and I understand an overhang to be

TexZeTech, 2023-08-25:
> **3mf File for This Bug**
> (Couldn't upload it here so I got it on WeTransfer)
> https://we.tl/t-zmw76rRNfm

Just compress to a .zip and you can upload to here.

celtare21, 2023-08-25:
> > **3mf File for This Bug**
> > (Couldn't upload it here so I got it on WeTransfer)
> > https://we.tl/t-zmw76rRNfm
> 
> Just compress to a .zip and you can upload to here.

thank you, updated

closed 2023-12-01 by github-actions[bot]