OrcaSlicer/OrcaSlicer #1882: P1S LAN mode not working

SurgeHawk, 2023-08-22:
**Describe the bug**
After slicing model and attempting to print on a P1S it states that IP or access code is wrong.

In the device tab Ocra can connect to the P1S normally, camera works normally. This issue only occurs when attempting to print.

**To Reproduce**
1 - Load model
2 - Slice model with any settings
3 - print model
4 - ask to verify IP/Access code, confirm
5 - states wrong IP/Access code

**Expected behavior**
To allow printing of sliced models with the correct IP/Access code

**Screenshots**
[image]

**Printer model**
Bambu Labs P1S

**Desktop (please complete the following information):**
 - Windows 10
 - Version 1.6.4 (Non-beta)

This issue was established with #1720 But has not had much traction.

SurgeHawk, 2023-08-23:
This is still an issue with the latest commit ff92578.

@SoftFever

gobau, 2023-08-24:
@SoftFever 
@SurgeHawk 

DeviceManager.cpp file
1392 lines
else if (printer_type == "C11")
=>
else if (printer_type == "C11" || printer_type == "C12")
When you change to , SSL is enabled and P1S LAN ONLY MODE operates normally.
If you give me the commit authority, I will modify it and request PR.

SoftFever, 2023-08-24:
> @SoftFever @SurgeHawk
> 
> DeviceManager.cpp file 1392 lines else if (printer_type == "C11") => else if (printer_type == "C11" || printer_type == "C12") When you change to , SSL is enabled and P1S LAN ONLY MODE operates normally. If you give me the commit authority, I will modify it and request PR.

@gobau Great. Looking forward 
To create a PR, here is the workflow: fork -> create branch from main for the fix -> PR

gobau, 2023-08-24:
@SoftFever 

PR Requested
https://github.com/SoftFever/OrcaSlicer/pull/1905

SurgeHawk, 2023-08-24:
@gobau 
@SoftFever 

This has been resolved in f7184c8 1.6.6

Thank you! Closing issue

closed 2023-08-24 by SurgeHawk

SoftFever, 2023-08-24:
cheer.
Thanks @gobau  for the fix.