OrcaSlicer/OrcaSlicer #1884: Keyboard shortcuts for duplicating object

Arakon, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
When wanting to print multiples of an object, it's quite annoying to have to right-click the object, select clone, select a number...
To speed this up, it would be great if the functionality from Superslicer could be added, where hitting + on the keyboard simply creates another clone of the currently selected object, without asking for an amount or any clicking needed.

**Describe the solution you'd like**
Functionality to press the + key to duplicate the currently selected object.

npoulos, 2023-08-26:
You can copy paste the object with the keyboard shortcuts ctrl/cmd-c ctrl/cmd-v for it (at least on Mac you can)

locki-cz, 2023-10-06:
> You can copy paste the object with the keyboard shortcuts ctrl/cmd-c ctrl/cmd-v for it (at least on Mac you can)

shortcut + is just better for object manipulation, paint seam on one object and you have it on all, but when you are not used to use it you cant understand that, when you copy that, you must paint seam on overy model manualy :)

closed 2024-01-12 by github-actions[bot]