OrcaSlicer/OrcaSlicer #1887: Internal Solid Infill affected by Top & Bottom Shell Layers

Durahl, 2023-08-22:
**Describe the bug**
I was trying to get rid of the _purple_ `Internal Solid Infill` and noticed it being affected by the `Top` & `Bottom Shell Layers`:
[image]
[image]
Obviously, with its generation being tied to the Top & Bottom Shell Layers one cannot edit them out like that without it conflicting with the actual Top & Bottom Shell Layers elsewhere needed on the Model 😑

**To Reproduce**
Load a Model with a fairly steep overhang value at or nearing 45° and have it Sliced with like 3 `Wall Loops` and 5 `Top` & `Bottom Shell Layers`.

**Expected behavior**
For a way to disable the generation of the `Internal Solid Infill` on `Overhangs` causing more unnecessary heat to get pumped into the local area.

**Screenshots**
See above

**Printer model**
[Voron V2.4 350](https://imgur.com/a/peDnYxr)

**Desktop (please complete the following information):**
 - OS: Windows 11 Professional
 - Version: 22H2

fir3bird, 2023-10-06:
I am having the exact same issue unfortunately.
This slicer has so much potential, but this and the internal bridge settings that cannot vary from other bridges are unfortunately two killer aspects why I am still not switching to Orca Slicer.

Apart from that, this is by far the most advanced slicer currently in my opinion.

PhilBaz, 2023-10-06:
Did you try disabling ensure vertical shell thickness?  Don't think this is a bug.

fir3bird, 2023-10-06:
Yes, I tried to disable "Ensure vertical shell thickness". Without any changes in preview nor printing time after reslicing. Maybe I should mention, that the models I print are no braindead standard geometries but tabletop terrain pieces.

PhilBaz, 2023-10-06:
@fir3bird and extra perimeters on overhangs?

Post a model so we can try and validate/invalidate.

fir3bird, 2023-10-06:
Extra perimeters on overhangs is disabled.
Unfortunately I am not allowed to share the exact model I am currently facing that issue with, but I think this one will do its job too:
https://www.myminifactory.com/object/3d-print-19-latchkey-row-ruin-275697

If you slice the `ground-floor.stl` you should see it near the chimney of the model.

PhilBaz, 2023-10-06:
One more thing try detect narrow solid infill!  I think that's just skin that is receiving a concentric pattern from having narrow corners. If its skin its supposed to be there.

fir3bird, 2023-10-06:
Unfortunately this also did not do the trick. Still the same result.

[image]

PhilBaz, 2023-10-06:
@fir3bird 

That's just skin dude! The model has voids below those. So it puts skin above them. maybe what you should be proposing is some minimum area filter. But that is expected behaviour.

fir3bird, 2023-10-06:
OK, seems like youre right. But there is still a significant difference between Cura and OrcaSlicer in printing time. Eventhough I tried to apply identical speed and accel. settings to both. Same infill patterns, wall and surface counts.

The time difference is around 1.25h for that little model. And the difference gap grows quickly with larger models.

Is there acutally a way to reduce this infill overshoot?

In a direct comparison Cura seems to infill a lot less area (Orca left - Cura right):
[image]

Thanks for your help so far! 😄

PhilBaz, 2023-10-06:
@fir3bird 

Well that also depends on how accurate your slicing estimates are. Out of the box Cura can be way off, not sure about Orca. Just moved to Orca from Crua recently.

That it the skin expand setting I was talking about. You can control the size of that skin patch. It's a good setting to have for sure. haven't seen anything like that in Orca.

Orca doesn't and probably won't ever have the level of control Cura has. Orca is nicer to use though. But for specific things ill still be using Cura.

PhilBaz, 2023-10-06:
@fir3bird PS....don't forget to close this issue.

gald to help!

fir3bird, 2023-10-06:
The estimates are very precise in both cases. Already printed things with Cura and Orca and their estimates have been very close to exact +/- 10 minutes at around 5h prints.

I really love orca slicer for that many features it brings out of the box, like multiple build plates and manual supports you can simply paint on the model. This are absolute killer features that let Cura stand in the rain. But that single point of that much longer printing time is something that bugs me. Apart from that just like I said, an awesome piece of software masterwork.

Would be really glad to see this feature in a future release. I can fully understand that orca tries to keep things simple, but optional settings for advanced users would be always welcome. :-)

Regarding that issue, I just hijacked it so I cannot close it - sorry.

Durahl, 2023-10-06:
Sorry about the late response - Only just today received a notification that there's actually a discussion going on regarding my issue 😑

Disabling `Ensure Vertical Shell Thickness` would _to some degree_ aid against the excessive `Internal Solid Infill` generation but not entirely - And most definitely not as cleanly as setting `Top/Bottom Shell Layers` to `0` which _cannot_ be the solution.

Same for `Extra Perimeters on Overhangs` which was already `disabled` by default in my settings.

As such I'd say the Problem is still not fixed in a satisfactory manner and as such should remain open because there's hardly a reason for _that much_ `Internal Solid Infill` to be generated.

PhilBaz, 2023-10-07:
@Durahl 

In both of your images 'ensure vertical shell thickness' is applied. Don't know if that's the issue, but wouldn't be able to tell from thos images. Also, at a certain overhang angle it may be considered skin, because it rightly should be, know what I mean?

You also have a numerical top shell thickness applied rather than just layer numbers which also may be impacting. From the pictures you posted the result is what you would expect...I believe.

ken830, 2023-11-03:
Not sure if what I'm seeing is related, but if I slice without top/bottom shells, the slicer still put in 1 layer of solid infill in mid-air for this overhang area. PrusaSlicer and Cura both put in sparse infill only, as expected.

[image]

This "solid infill" still follows the solid infill pattern setting (set pattern before setting shell to zero) even though it is greyed-out when the bottom shell setting is set to zero. This seems to indicate that this is not expected behavior.

[image]

I had to workaround this by using a local modifier to enable 1 bottom shell so that it will create a solid, straight bridging lines here to support the sparse infill of subsequent layers.

PR #3235 (2024-01-18): [Feature] Introduced experimental option to fully disable solid infill against walls when ensure vertical thickness is turned off
Introduced experimental option to fully disable solid infill printed against sloping walls even when ensure vertical thickness is turned off.

Fixes #3224
Fixes #3016
Fixes #1887

**Ensure vertical thickness ON:**
We can see solid infill against the walls even on pretty much flat surfaces. This is the most conservative option.
[image]

**Ensure vertical thickness OFF:**
Solid infill greatly reduced in mostly flat areas. 
[image]

However it is present on sharp slopi [... cut]
closed 2024-01-18 by SoftFever