OrcaSlicer/OrcaSlicer #1888: Max flowrate calibration print not TPU compatible (P1P)

PrincessLuna2253, 2023-08-23:
**Describe the bug**
When performing the max flowrate calibration print, the slicer sets the max volumetric speed for the selected filament to 200mm3/s, which causes two problems for filaments with very low max volumetric flow values like TPU. 

The first is that the starting gcode for the P1P uses the volumetric flow limit to calculate how fast to print the priming line. This resulted in my P1P attempting to print the priming line at 15mm3/s which caused a jam. 

The second is that now the first layer prints at full speed, 50mm/s for me, which isn't as fast as the priming line but was still above 5mm3/s which was enough to cause the filament to jam.

**To Reproduce**
Generate a calibration print with appropriate values for TPU (1mm3/s start, 4mm3/s end, 0.2mm3/s step in my case), check the flow rates on the preview tab

**Expected behavior**
The volumetric flow limit should be set to the chosen end value instead of 200mm3/s, this would prevent the issue as long as a sensible end value is chosen

**Screenshots**
[image]

**Printer model**
Bambu Labs P1P

SoftFever, 2023-08-24:
We'll need to set a reasonable max volumetric flow in the calibration dialog ourselves.
We can't rely on the max volumetric flow in the filament setting, as that would defeat the purpose of this test.

PrincessLuna2253, 2023-08-24:
If the flow rate limit is automatically set to the max flow rate needed for the generated test it shouldn't interfere, and as long as a reasonable max value is chosen by the user it should make it so the filament will likely underextrude on the priming line and first layer but wouldn't strip. That said, adding max flow rate to the calibration dialog also sounds like a perfectly good solution.

closed 2023-12-01 by github-actions[bot]