OrcaSlicer/OrcaSlicer #1894: Allow more formats of G-Code Thumbnails as Prusa Slicer

frasier2k, 2023-08-23:
Please implement the ability to select the format of the G-Code thumbnails.

[image]

PanelDue only supports QOI format.

discip, 2023-08-24:
Would be nice if the [BIQU tumbnail](https://github.com/bigtreetech/Bigtree3DPluginSuit) option would be implemented as well.

freiser77, 2023-09-11:
I would like the QIDI thumbnail option.

discip, 2023-10-07:
Would be very nice if we could get this format [BIQU tumbnail](https://github.com/bigtreetech/Bigtree3DPluginSuit) supported as well.

@effgarces maintains a python script ([Biqu-Thumbnail-Generator](https://github.com/effgarces/Biqu-Thumbnail-Generator)) to achieve that.

But unfortunately the resulting pics are only about 1/2 of the size compared to the thumbnails generated by the Cura plugin.

## Cura Plugin
```
;00460046
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000086118e3296531a639e739e739e731a6294518e3084100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000861318652aa73ae84308c518c71947194929492949294718c718c518430738e528a296508410000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000021245acb8430949194b29cf3a513a514a534ad34ad54ad55ad54ad34a534a514a4f39cd394b28c71841052aa18e300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000000018e3632c8c519cb2a513a534ad55ad75b596b5b6b5b6bdd6bdd7bdd7bdd7bdd6b5b6b5b6b596ad75ad55a5349cf394b28c505aeb10a2000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000204a49843094b2a514ad55b596b5b6bdd7bdf7c617c618c638c638c638c638c638c638c638c618c5f7bdf7bdd7b5b6b595ad55a51494b2841039e700000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000001082632c8c719cf3ad55b596bdd7c5f7c618c638ce58ce59ce79ce79d699d699d699d699d679ce79ce79ce59ce58c638c618bdf7bdd7b596ad559cf38c515acb0841000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000010a26b6d9492a514b575bdd7c617ce38ce59d679d69ad6bad6badedadedadedbdedbdedbdedbdedbdedadebad6bad6bad69ace79ce59c638c617bdd6ad75a5138c71632c084100000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000008616b6d9491a514b596bdf7c638ce79d69adebadedbdefbe71be71ce71ce73ce73ce73ce73ce73ce73ce73ce71ce71ce71bdefbdedbd6bad69ace59c638bdd7ad75a5148c71630c00200000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000205aeb8c51a514b595bdf7ce38d699d6badefbe71be73cef5cef5def7def7df77df79df79df79df79df79df77def7def7def5def3ce73ce71bdedbd6bad679c638bdd7ad759cf38c5052890000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000039e784109cf3ad75bdd7c638d699dedadefbe73cef5cef7df79df79ef7bef7beffdeffdeffdeffdfffdfffdeffdef7bef7bef79ef77def7def5ce73cdefbdebad679c638bdd7ad559cd37bef2965000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000010a273ae9492a534b5b6c638ce79debae6fbe73cef5df79df7beffdeffdfffffffffffffffffffffffffffffffffffffffffffffffdfffbef79ef77def5de73cdefbdebace79c618b5b6a53494716b6d084100000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000004a6984309cf3b575bdf7ce59d6badefbe73cef5df79df7beffdfffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffdff7bef79def5de73cdefbd6bace58bdf7ad759cd3841039e700000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000010a2736d9471a534b5b6c638d699dedbe71cef5df79df7beffdfffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffdff7bef77def5ce71bdedace79c618b5b6a5148c716b4c0841000000000000000000000000000000000000000000000000 [... cut ...]
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
; bigtree thumbnail end
```

There has been a lot going on regarding this topic for the most part in SuperSlicer:
https://github.com/supermerill/SuperSlicer/issues/887

## Maybe someone could finally get this going by implementing it in the best slicer ever. 😊
@Noisyfox @Ocraftyone @igiannakas

Ocraftyone, 2023-10-11:
@discip I'm assuming you have one of the BTT since you are requesting it. If you had a moment and wouldn't mind trying some g-code for me, that would be nice. I have an implementation thrown together, but the output is not identical (just seems like the coloring is different, but this could be because I am going directly from the raw RGBA values rather than RGBA >> PNG >> BTT format). This format also includes all sizes that the original BTT plugin does.

https://pastebin.com/nCNJeMSg

Ocraftyone, 2023-10-11:
@frasier2k I have also implemented the option to choose the format to save in as well as adding JPG and QOI formats. I am waiting to see if the BTT format works so I can merge them all together

discip, 2023-10-11:
Will test as soon as possible.

discip, 2023-10-11:
Good evening Sir,
I'm so pleased to see someone getting this finally going!

# THANK YOU SO MUCH! 😃👍🎉

Unfortunately what you provided did not work. 😔

There is one little but effective quirk with the size of the generated thumbnails.
I'm using a **`TFT28`** and a **`TFT43`**. The thumbnails on the TFT28 are really small when generated with the converter I linked above.
The thumbnails generated with the cura-plugin crop everything 'black away' while preserving the aspect ratio. (I don't know if it actually crops, but at least it uses all the available space.)

That's the reason I uploaded these samples above.
Hope you could make it generate thumbnails the _cura-plugin-way_. 😅

I'm willing to help where ever I can.

Additionally there is something very interesting going on for different resolutions.

discip, 2023-10-11:
Good evening, sir,
I'm so happy that someone is finally taking care of this!

# THANK YOU SO VERY MUCH! 😃👍🎉

1. Unfortunately the sequence you shared didn't work. 😔

2. There is a small but noticeable quirk regarding the size of the generated thumbnails.
I use a **`TFT28`** and a **`TFT43`**. The thumbnails on the **`TFT28`** are really small when generated using the converter linked above.
The thumbnails generated with the _Cura-Plugin_ crop everything “black” while maintaining the aspect ratio. (I don't know if it actually crops, but at least it uses all the available space for the image.)

   That's why I uploaded these examples above.
I hope you manage to get thumbnails generated the Cura-Plugin-way. 😅

I'm willing to help where ever I can.

3. Furthermore, there is something very interesting going on for different resolutions.
e.g.:
- **`TFT28`** only requires the 1st block (70x70)
- **`TFT43`** requires the 1st, 2nd and 3rd block (70x70, 95x80, 95x95)

Unfortunately I only have these 2 TFTs.
Otherwise I would have tested the other ones too. 😁

I'm excited to see how this will turn out in the end.

Thank you!
kind regards

Ocraftyone, 2023-10-11:
Hmm. That's no good. Looking at superslicer, there is an implementation in there. Have you tried that implantation recently?

discip, 2023-10-11:
Good evening, sir,
> @discip I'm assuming you have one of the BTT since you are requesting it. If you had a moment and wouldn't mind trying some g-code for me, that would be nice.

I'm so happy that someone is finally taking care of this!
Of course I am willing to try what you linked!

# THANK YOU SO VERY MUCH! 😃👍🎉
1. 
   > I have an implementation thrown together, but the output is not identical (just seems like the coloring is different, but this could be because I am going directly from the raw RGBA values rather than RGBA >> PNG >> BTT format)

   Unfortunately the sequence you shared didn't work. 😔

2.
   > This format also includes all sizes that the original BTT plugin does.

   There is a small but noticeable quirk regarding the size of the generated thumbnails.
I use a **`TFT28`** and a **`TFT43`**. The thumbnails on the **`TFT28`** are really small when generated using the converter linked above.
The thumbnails generated with the _Cura-Plugin_ crop everything 'black'/'transparent' while maintaining the aspect ratio. (I don't know if it actually crops, but at least it uses all the available space for the image.)

   That's why I uploaded these examples above.
I hope you manage to get thumbnails generated the Cura-Plugin-way. 😅

I'm willing to help where ever I can.

3. Furthermore, there is something very interesting going on for different resolutions.
e.g.:
- **`TFT28`** only requires the 1st block (70x70)
- **`TFT43`** requires the 1st, 2nd and 3rd block (70x70, 95x80, 95x95)

Unfortunately I only have these 2 TFTs.
Otherwise I would have tested the other ones too. 😁

I'm excited to see how this will turn out in the end.

Thank you!
kind regards

Ocraftyone, 2023-10-11:
Also, after looking through the cura plugin a bit, it seems that what resizes the view of the picture is cura, not the plugin.

discip, 2023-10-12:
Do you think it's possible to get Orca to do this in a comparable way?

discip, 2023-10-12:
Do you think it's possible to get Orca to do this in a comparable way?

Would you mind sharing the 3mf file with the cube you used so I can compare the results?

discip, 2023-10-12:
update:

Today I got the chance to test your code on my TFT28 and for some reason on that one it works.
I have an idea to test as soon as the print on the printer with the TFT43 has finished.

Ocraftyone, 2023-10-12:
hmm. thats interesting. do you think maybe the extra resolutions in between may be causing an issue?

Ocraftyone, 2023-10-12:
im also going to throw this here for reference - from BTT firmware repo

    70x70: TFT24 / TFT28
    95x80: TFT43 / TFT50
    95x95: TFT35
    160x140: TFT70

discip, 2023-10-12:
My idea was to test if it will work when the actual object reaches up to the borders, like in the example I initially provided.
Unfortunately I will have to wait until another print is finished. 😅

discip, 2023-10-12:
One more thought about the size.
Where does the internal converter get the raw image get from. Is it the preview in the upper left corner?

[image]

If so, is it possible to change the size for the preview?

Ocraftyone, 2023-10-12:
ok. i am taking a different approach and looking into the source code of the tft itself and seeing how it converts the code and I found the function it uses. Now going to look into how it loads the image data.

Ocraftyone, 2023-10-12:
So just to confirm, if you take a gcode file with png and run it through the converter you provided, it does work right?

discip, 2023-10-12:
I am only using it within Orca as a post processor.
I did not try to use it outside of Orca.

Ocraftyone, 2023-10-12:
that is fine. but it is working right?

discip, 2023-10-12:
Yes it is for both the MKS28 & TFT43.

Ocraftyone, 2023-10-12:
it seems as there is a certain memory offset that it seeks to if it does not find the header specified above. Im going to try and hack this together and see if I can get it working

discip, 2023-10-12:
Could you provide me with the cube.3mf file You used to generate the gcode please?

Ocraftyone, 2023-10-12:
yeah, it is the primitive cube in orca. just right click on the plate and select add

discip, 2023-10-12:
Could you please also provide me the exact color of the cube?
Since the filament I set up is white, while yours was orange.

discip, 2023-10-12:
You seem to be right about the offset, at least that is what I think, because depending on which resolution your TFT needs, which in turn is arranged the way you listed it above, it needs all the resolution prior to that to work.
e.g.:
If you happen to have a FTF43 which has a resolution of 95x80 you need not only the 2nd block in the g-code file but also the 1st block as well even if it is completely filled with **`0`**!

As mentioned above:
> * **`TFT28`** only requires the 1st block (70x70)
> * **`TFT43`** requires the 1st, 2nd and 3rd block (70x70, 95x80)

Ocraftyone, 2023-10-12:
Color of the cube is 255, 128, 64 (r, g, b)

And I also have another one to try:

```
;00460046
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000008200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba738e1000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7820400000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba70820000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba738e100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000008208204eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba782040820000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba738e100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba782040000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7082000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba738e10000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000000000820eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba78204082000000000000000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000000000000000038e1eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba738e10000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000008204eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba7eba78204082000000000000000000000000000000000000000000000000000000000000000000000
[... cut ...] 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
; bigtree thumbnail end

```

discip, 2023-10-12:
There is something really strange going on!
If I slice the cube with the post processor enabled and replace the created data with yours it works, but as soon as I slice another object and replace the existing data with yours, it doesn't. 🤷🏻‍♂️

Ocraftyone, 2023-10-12:
Was this while trying the one I just sent?

discip, 2023-10-12:
Yes!

discip, 2023-10-12:
I just retested and it worked with your newly sent code.
Sorry for the misinformation.

Ocraftyone, 2023-10-12:
WOOOOOHOOOOOO!

Ocraftyone, 2023-10-12:
[Cube_PLA_25m57s.zip](https://github.com/SoftFever/OrcaSlicer/files/12887598/Cube_PLA_25m57s.zip)
Wanna double check with this gcode file and make sure?

discip, 2023-10-12:
Unfortunately this does for what ever reason not work! 😭

Ocraftyone, 2023-10-12:
🤦 
Are you able to send me the gcode file that did work?

discip, 2023-10-12:
So I tried further and noticed, that If you slice the file with the post processor enabled and than replace the blocks of code representing the thumbnail with your code even with the the 1st one you sent, it works! 🤷🏻‍♂️

> Are you able to send me the gcode file that did work?

Will do.

discip, 2023-10-12:
[cephalopo_II.zip](https://github.com/SoftFever/OrcaSlicer/files/12887666/cephalopo_II.zip)

discip, 2023-10-12:
I think I found the culprit:
[image]

In the **working** gcode every line ends with this interesting shaped cirlce.

While in the **none working** file every line ends with this reversed 'P'.

Ocraftyone, 2023-10-12:
it definitely has to do with the end of line characters. the problem is that when the slicer exports the gcode file, my line endings are converted back. Im going to attach a file that attempts to fool the firmware by providing spaces rather than the extra end of line character and hope that works. after looking at the firmware code again, this should work as all it is doing is reading 4 characters at a time. with each end of line character really being 2 characters each, it would make each line divisible by 4 characters rather than just having the one end of line character causing that remainder

Ocraftyone, 2023-10-12:
[Cube_PLA_25m58s.zip](https://github.com/SoftFever/OrcaSlicer/files/12887752/Cube_PLA_25m58s.zip)

Ocraftyone, 2023-10-12:
> I think I found the culprit: [image]
> 
> In the **working** gcode every line ends with this interesting shaped cirlce.
> 
> While in the **none working** file every line ends with this reversed 'P'.

Yup! Beat me by a second lol. The P indicates a "\n" in the file while the square/circle indicates "\r\n" (which is what we want). hopefully it likes "  \n" just as well 😜

discip, 2023-10-12:
Nope, that did not do the trick! Sigh

discip, 2023-10-12:
But how is this achieved.
If I understand this right the file itself must be formatted in some way, as the code added after the file has been generated also ends with the desired "\r\n". 🤷🏻‍♂️

discip, 2023-10-12:
Maybe the answer lies in this python script?
[biqu_convert_p24.zip](https://github.com/SoftFever/OrcaSlicer/files/12887812/biqu_convert_p24.zip)

discip, 2023-10-12:
This is the part of the code containing `\r\n` 4 times.
Maybe this helps. 😊

```python
def overread(msize,gfile):
    moutdata = ""
    img = QImage()
    img.loadFromData(base64topng(readPSTump(gfile),''))
    img = img.scaled(msize.width(),msize.height(), Qt.IgnoreAspectRatio, Qt.SmoothTransformation)
    moutdata +=  ";"+(hex(msize.width())[2:]).rjust(4,'0')+(hex(msize.height())[2:]).rjust(4,'0')+"\r\n"
    pos = QSize(0,0)
    for ypos in range(0,img.height()):
     qrgb =";"
     for xpos in range(0,img.width()):
       data = img.pixel(xpos,ypos)
       pos.setWidth(pos.width()+1)
       dummy = (hex(((data & 0x00F80000) >> 8 ) | ((data & 0x0000FC00) >> 5 ) | ((data & 0x000000F8) >> 3 ))[2:]).rjust(4,'0')
       if dummy == "0020" or dummy == "0841" or dummy == "0861":
         dummy = "0000"
       qrgb = qrgb + dummy
     pos.setWidth(0)
     pos.setHeight(pos.height()+1)
     moutdata = moutdata + qrgb + "\r\n"    
    return moutdata

def overseek(gfile):
        outdatar = ""
        outdatar = outdatar + overread(QSize(70,70),gfile)
        return outdatar

def do_convert(gfile):
       env_slicer_pp_output_name = str(os.getenv('SLIC3R_PP_OUTPUT_NAME'))
       final_name = os.path.splitext(env_slicer_pp_output_name)[0]+os.path.splitext(env_slicer_pp_output_name)[1]
       with open(gfile + '.output_name', mode='w', encoding='UTF-8') as fopen:
           fopen.write(final_name)
       outdata = ""
       outdata = overseek(gfile)
       outdata = outdata + "; bigtree thumbnail end\r\n\r\n"
       fh = QFile(gfile)
       fh.open(QIODevice.ReadOnly)
       stream = QTextStream(fh)
       stream.setCodec(CODEC)
       lino = 0
       fg = stream.readAll()
       fh.close()
       bigtree3dfile = gfile
       fh = QFile(bigtree3dfile)
       fh.open(QIODevice.WriteOnly)
       stream = QTextStream(fh)
       stream.setCodec(CODEC)
       stream << outdata
       stream << fg
       fh.close()
```

Ocraftyone, 2023-10-12:
there is something withing the rest of the code that is actively converting \r\n to \n. im looking through the file write process rn but I may need to pick this up later. at least were on the right track of what is working and what isn't

Ocraftyone, 2023-10-13:
Got it! Some post processing scripts run after to add in things like time remaining and other things that have to be calculated after the main gcode has been fully written and instead of reusing the end of line character present in the original file, it just uses \n at the end of each line. Got it patched so that it now determines the format previously used and keeps it original.

[Cube_PLA_25m57s.zip](https://github.com/SoftFever/OrcaSlicer/files/12892899/Cube_PLA_25m57s.zip)

discip, 2023-10-13:
Amazing!

# Thank you very much!

After this being sorted out, do you think you will be able to get the Thumbnail bigger?

Ocraftyone, 2023-10-13:
I would be happy to take a look, but I will do that in a different pull request. I already have a decent amount in this one. Just to confirm, did the last gcode file work?

discip, 2023-10-13:
Yes it definitely did.
But I noticed that there were 3 picture generated:
70x70
95x80
95x95

I think it would be best, if only the necessary blocks were generated:
resolution|TFT|wich gode-blocks
:-:|:-:|:-:
70x70|TFT24 / TFT28|1st
95x80|TFT43 / TFT50|1st & 2nd
95x95|TFT35|1st, 2nd & 3rd
160x140|TFT70|1st, 2nd, 3rd & 4th

discip, 2023-10-13:
# Yes, it definitely did. 👍🏻
But I noticed that there were 3 code-blocks (I think you know what I mean by that 😊) generated:
70x70
95x80
95x95

I think it would be best, if only the necessary blocks were generated depending on what is entered in the corresponding field:
resolution|TFT|wich gode-blocks
:-:|:-:|:-:
**70x70**|TFT24 / TFT28|1st
**95x80**|TFT43 / TFT50|1st & 2nd
**95x95**|TFT35|1st, 2nd & 3rd
**160x140**|TFT70|1st, 2nd, 3rd & 4th

discip, 2023-10-13:
# Yes, it definitely did. 👍🏻
But I noticed that there were 3 code-blocks (I think you know what I mean by that 😊) generated:
70x70
95x80
95x95

I think it would be best, if only the necessary blocks were generated depending on what is entered in the corresponding field:
resolution|TFT|necessary gode-blocks
:-:|:-:|:-:
**70x70**|TFT24 / TFT28|1st
**95x80**|TFT43 / TFT50|1st & 2nd
**95x95**|TFT35|1st, 2nd & 3rd
**160x140**|TFT70|1st, 2nd, 3rd & 4th

Ocraftyone, 2023-10-13:
Yeah. I was thinking about having a drop down option replace the thumbnail sizes option whenever BIQU is selected. My only concern with that would be if BIQU ever came out with a new display, the source code would need to be modified rather than just user settings

discip, 2023-10-13:
What about making the drop-down dependent on a specific user setting.

Ocraftyone, 2023-10-13:
Possibly. Another option would be adding a "manual" option to the dropdown

discip, 2023-10-13:
You are the one in the know here.
What ever will make it work while keeping it easy to maintain. 😊

Do as you wish! 😀

discip, 2023-10-13:
btw:
Regarding the thumbnails size improvement:

> I would be happy to take a look, but I will do that in a different pull request.

Do I need to open a dedicated issue regarding this or will you make a PR either way?

Ocraftyone, 2023-10-14:
I am working on an implementation rn. Currently having an issue where the config value keeps resetting to default in one of the places I am trying to read it.

[1 more reply]

PR #2405 (2023-10-17): Thumbnail Formats Option Ported from PrusaSlicer and add BIQU/BTT format
First major change is adding the option for thumbnail format under the thumbnail size option. The options ported from PrusaSlicer are PNJ, PJG, and QOI.
Second major change is the addition of the BIQU, or BigTreeTech, image format that is used on their TFTs. Doing this also required a small change to the GCodeProcessor.post_process method that passes through the existing end of line characters

fixes #1894
closed 2023-10-17 by SoftFever