OrcaSlicer/OrcaSlicer #1898: Pattern method generate invalid line gcode

DiomedesDominguez, 2023-08-23:
**Describe the bug**
For a Prusa MK3S+ it's generating and odd gcode to exclude a block.

```EXCLUDE_OBJECT_DEFINE NAME=Orca-PA-Calibration-Test CENTER=0,0 POLYGON=[[5,5],[245,5],[245,205],[5,205],[5,5]]```

This line make the Prusa to restart every single time. After removing it, it was fixed.

**3mf File for This Bug**
gcode file [here](https://mega.nz/file/3V8CBS7b#vNzF8-DMdNlK-2t-P1tK5EUyg49B_Mb1zybXdEUpbPQ)

**To Reproduce**
Slice the Pattern method for a Prusa MK3S+, export the file (SD or octoprint doesn't matter) try to print the file

**Expected behavior**
start printing

**Printer model**
Prusa MK3S+ 

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.6.4

SoftFever, 2023-08-24:
This is generate for Klipper's the object exlusion feature.
Did you change MK3S's gcode style to Klipper accidently?
Change back to Marlin, this won't be generated.

DiomedesDominguez, 2023-08-24:
No I don't change it to Klipper, I double check it.

DiomedesDominguez, 2023-08-27:
Here is a video using the version1.6.6
https://youtu.be/2UQAsmex3wk

SoftFever, 2023-08-27:
ah, I see
Yes, it's a bug...

SoftFever, 2023-08-27:
Fxied in dev branch

closed 2023-08-27 by DiomedesDominguez