OrcaSlicer/OrcaSlicer #1901: 1.6.4 Missing System Profiles and User Profiles

jbeardon, 2023-08-23:
**Describe the bug**
A number of system profiles appear to be missing in 1.6.4
Eg, There used to be a system profile for the Creality Ender 5 with .6 nozzle in 1.6.3 there is only a .4 nozzle version in 1.6.4
This appears to have a knock on effect that prevents any user profiles that were based upon the missing system profiles from appearing in the user profiles list

**3mf File for This Bug**
NA

**To Reproduce**
Open slicer - create project - try to select either a printer, print settings or filament profile - profiles are not present.

**Expected behavior**
I expect the profiles to be there!

**Screenshots**
NA

**Printer model**
Bambu P1P
Creality K1 Max with .6 nozzle - user profile based off the Ender5 Plus with .6 nozzle

**Desktop (please complete the following information):**
Windows 11

SoftFever, 2023-08-24:
can you try the 1.6.6 see if that still has the issue?

jbeardon, 2023-08-24:
Afraid so
The .6 nozzle versions of creality printers are not available to add as system printers and my custom presets based on the .6 versions are still unavailable.
Screenshots below

[image]
1.6.3

[image]
1.6.6

[image]
1.6.3 (checked preset is the one - ignore the others they are me trying to work-around the issue - this was based on the .6 Ender 5 and is missing from 1.6.6), as are the filament presets that were based on the default presets for the .6 E5+

SoftFever, 2023-08-24:
Hmm, that’s strange.I don’t think 1.6.3 has 0.6 version.Did you copy it from somewhere else by yourself?You probably need to copied that to the 1.6.6 again. Another workaround is, you can find your custom profile(the .json file) in “%appdata%/orcaslicer/user” folderOpen it with notepad, find the "inherits" key and change it’s value to the 0.4 nozzle profile:Sent from Mail for Windows From: jbeardonSent: Thursday, 24 August 2023 9:45 pmTo: SoftFever/OrcaSlicerCc: SoftFever; CommentSubject: Re: [SoftFever/OrcaSlicer] 1.6.4 Missing System Profiles and User Profiles (Issue #1901) Afraid soThe .6 nozzle versions of creality printers are not available to add as system printers and my custom presets based on the .6 versions are still unavailable.Screenshots below1.6.31.6.61.6.3 (checked preset is the one - ignore the others they are me trying to work-around the issue - this was based on the .6 Ender 5 and is missing from 1.6.6), as are the filament presets that were based on the default presets for the .6 E5+

jbeardon, 2023-08-24:
Ah....
Damn and blast. I think it may have been included when I was attempting to add K1 Max profiles from GalaxySlicer (which, incidentally, were crap). My bad. Sorry.

closed 2023-08-24 by jbeardon