OrcaSlicer/OrcaSlicer #1904: "Fill bed with copies" does not perform well (enough)

foreachthing, 2023-08-24:
## Describe the bug
Comparing to PrusaSlicer, OrcaSlicer does not perform very well in "filling bed" (I only wish those where instances over copies - but that's another (open) issue).

## 3mf File for This Bug
No need, STEP file is attached.
[USM-Headset.zip](https://github.com/SoftFever/OrcaSlicer/files/12426053/USM-Headset.zip)

## To Reproduce
1. load STEP
1. check "arrange" settings to be: 3.5 mm distance, rotation enabled (<- if this matters at all?!).
1. "Fill bed with copies"
1. observe results
1. repeat with PrusaSlicer (>= 2.6.0)

## Expected behavior
Very similar results, if not the same.

## Screenshots
### OrcaSlicer
With rotation enabled:
[image]

With rotation disabled 1:
[image]

With rotation disabled 2:
[image]

### PrusaSlicer
[image]

## Desktop
- Windows 10/11
- OrcaSlicer 1.6.4
- PrusaSlicer 2.6.1rc1

Noisyfox, 2023-08-24:
Not sure if it's caused by the "Arrange improvements" added in https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1 have you tried PS 2.6.0?

foreachthing, 2023-08-24:
Performance in 2.6 is the same.
Screenshot from 2.6.0 (arrange options: distance 3.5 mm, 1 mm spacing to plate and rotation enabled):
[image]

So, I guess, it does not matter whether you port 2.6 or 2.6.1 ;-) If _I_ could make a wish, I'd like to be able to nest within objects => 2.6.1rc1 _(I yet have to try - but I want to keep using OrcaSlicer ...)_.

Noisyfox, 2023-09-20:
[image]
Much better now.

foreachthing, 2023-09-21:
Not convinced, on my system 😆 
[image]
To be fair, this was with a custom bed shape.

And now, with a rectangle:
_Naaa... same difference._
[image]

Noisyfox, 2023-09-21:
Ah I didn't mean the production version. It's still under development and (mostly) working on my own dev branch.

foreachthing, 2023-12-21:
This has yet to be fixed: "Fill bed with copies"
[image]

kleinwareio, 2024-03-21:
Still an issue with round build plates

foreachthing, 2024-03-22:
[image]

With rectangular plates too.

foreachthing, 2024-06-21:
This looks niiiiice, in 2.1rc1!!!
[image]

closed 2024-06-21 by foreachthing