OrcaSlicer/OrcaSlicer #1911: Order of inner wall/outer wall/infill modifier not working properly

user32rtu, 2023-08-24:
**Describe the bug**
On the inside of the model, the "Order of inner wall/outer wall/infill" modifier does not work correctly.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
1. The perimeter printing order is globally set to "inner wall/outer wall/infill".
2. To change the order of printing perimeters in a certain part of the model, a modifier in the form of a cylinder has been added.
3. For the modifier, the order of printing perimeters was set to "Order of inner wall/outer wall/infill".

**Expected behavior**
The order of printing perimeters for the part of the model to which the modifier is applied should be "outer wall/inner wall/infill".

**Screenshots*
[image]
*
[image]
[image]
[image]
[image]

**Printer model**
SapphirePro

**Desktop (please complete the following information):**
 - OS: Windows
 - Version Windows 10 Pro 22H2

GKSN, 2024-11-28:
Having the same problem here, the only difference is that I'm trying to use the inner/Outer on the infill... the slicer only consider the global settings and not the modifier.

zeno-tek, 2025-02-15:
This feature does work, technically, if your modifier covers an entire layer.  Once you have some parts using inner/outer and others using inner/outer/inner _on the same layer_, it defaults to the global settings.