OrcaSlicer/OrcaSlicer #1913: Unable to edit Auxillary Fan Speed on filament profiles.

Soygen, 2023-08-24:
When editing a custom filament profile, the aux fan speed setting is grayed out and I'm unable to edit it. I'm able to edit and save the setting in Bambu Studios, however. See picture:

[image]

This is on Windows 10 and this started happening to me on a beta version of 1.6.4 but it's still  happening on 1.6.6 Any help is appreciated.

SoftFever, 2023-08-25:
Hi
You will need to enable aux fan in machine profile

Soygen, 2023-08-25:
> Hi You will need to enable aux fan in machine profile

Thanks so much. I must've missed that change when it went in. Thanks for all the work you do.

closed 2023-08-25 by Soygen