OrcaSlicer/OrcaSlicer #1915: OS seems to have hard time to deal with large files

R4toLoko, 2023-08-25:
**Describe the bug**

I am trying to print this model just like the usual: https://www.printables.com/model/301559-rope-bowl
The model is taking easily close to if not more than 2-3min to slice, nothing wrong with that when it involves complex meshes.

When I try to send it to my Voron tho, nothing happens, but here is the problem.
I can hear my laptop ( Ryzen7, 32GB, RTX3050Ti, NVMe ) cooling system becoming a helicopter so things are happening in the background. Any interaction asks you to foce quit OrcaSlicer so something is not right.
I had to export as gcode and manually upload it to Mainsail

The g-code file is around 110MB

Export g-cgode- file works just fine so the file is ready, no idea why it crashes to upload it.
In theory, grab the very same file and upload it, no need to process anything that explains CPU usage going so high triggering the laptop cooling system.

Due to the fact that the cooling system goes mad, something is telling me that the process to upload the file is somewhat getting lost or need to be optimized.
If you keep trying, you get some Python errors which I failed to screenshot.

[image]

**To Reproduce**

1. Slice the model as usual
2. Hit "Upload" or "Upload and Print" and nothing happens

**Expected behavior**
Upload the file to the printer without crashing the slicer.
This is a local only printer, the upload via Mainsail works just fine, there is no network issue.

**Printer model**
LDO Voron Trident

**Desktop (please complete the following information):**
 OS: Linux Mint Cinammon 21.1

Thank you

Noisyfox, 2023-08-25:
> If you keep trying, you get some Python errors which I failed to screenshot.

Did you set up a postprocess script?

R4toLoko, 2023-08-25:
> > If you keep trying, you get some Python errors which I failed to screenshot.
> 
> Did you set up a postprocess script in python?

A-ha!
That makes sense.
Thank you.

closed 2023-08-25 by R4toLoko