OrcaSlicer/OrcaSlicer #1919: Crash at finish of print "running out of memory" - "bad allocation"

dartrax, 2023-08-25:
**Describe the bug**
After the print finished I was greeted with this message:

> Fatal error
> 
> OrcaSlicer will terminate because of running out of memory.It may be a bug. It will be appreciated if you report the issue to our team.
> 
> bad allocation

**To Reproduce**
It happened while OrcaSlicer was idle in the background, showing the device page, Camera stream not loaded. The printer reached 100%. Not sure if that was coincidence. I'm also not sure if this can be reproduced. I'll report here once it happens again.

**Screenshots**
[image]
[image]

**Printer model**
Bambu X1C

**Desktop (please complete the following information):**
 - OS: Win 10 64bit

bergmannm, 2023-09-24:
Same issue here with latest 1.7.0 release. Is there somewhere on my system stored a crashlog with more detailed information about what happened?

bergmannm, 2023-09-24:
There is definitely a bug, which keeps eating memory. Looking at memory stats for Orca Slicer - active private working set - I can see, that Orca keeps eating memory at about 5 MB per second.

tomlutzenberger, 2023-10-26:
Same issue here, also like @dartrax on Win10 64Bit and with Bambulab X1C. I'm experiencing it since v1.7.0.
I don't think it has to do with the device view + connected Bambulab, as it also happenend while being in prepare and preview view and before I got the X1C. Sometimes it crashes only a few minutes after starting OrcaSlicer, sometimes it takes hours. I wasn't able to take screenshots or look into task manager as it even leads to screen flickering and crashing the entire computer (without blue screen). Only rebooting helps.

SirThunder88, 2023-11-19:
Also having that issue, win10 64b i7 32G of ram.  Usually happens after having it open for a day or so, version 1.8.0 beta 2.  
[image]

Had a similar issue to @tomlutzenberger but found that an HD sound process was using a lot of cpu for no discernible reason; once I killed it I could start orca slicer again and the weird behaviors cleared up.

dartrax, 2023-12-07:
I had that issue again with current v1.8.1
Orca was idle in background for several hours.

Zechs2205, 2023-12-18:
I have the same issue

Mickelin01, 2023-12-23:
Same issue, both 1.8.0 and 1.8.1 on Win11

tomlutzenberger, 2023-12-24:
Update from my side: It stopped happening after I upgraded my RAM from 16 to 32GB in early November. I don't know if this could be because of some corrupted sections of the old RAM or if OrcaSlicer was just eating up a lot of memory occasionally. I didn't monitor the behavior a lot but it seems that it stopped altogether. I can let OrcaSlicer run in the background all day now without issues.
I'm still running v1.7.0.

silence48, 2023-12-31:
I don't see any logs on this one, but the issue seems to be the same as what's goin on with all the others that have the Exception code: 0xc0000005
Those related issues i've found includes: 
[2921](https://github.com/SoftFever/OrcaSlicer/issues/2921) 
[3375](https://github.com/SoftFever/OrcaSlicer/issues/3375) 
[3157](https://github.com/SoftFever/OrcaSlicer/issues/3157)
[3340](https://github.com/SoftFever/OrcaSlicer/issues/3340)
[3292](https://github.com/SoftFever/OrcaSlicer/issues/3292)
[2896](https://github.com/SoftFever/OrcaSlicer/issues/2896)
[2766](https://github.com/SoftFever/OrcaSlicer/issues/2766)
[2888](https://github.com/SoftFever/OrcaSlicer/issues/2888)
[3174](https://github.com/SoftFever/OrcaSlicer/issues/3174)
[2803](https://github.com/SoftFever/OrcaSlicer/issues/2803)
[2484](https://github.com/SoftFever/OrcaSlicer/issues/2484)
[2384](https://github.com/SoftFever/OrcaSlicer/issues/2384) -- This one says it was resolved yet it's logs reflect it being the same issue.
[2268](https://github.com/SoftFever/OrcaSlicer/issues/2268)

dartrax, 2023-12-31:
Why do you think so?
Looking at some of the issues you‘ve linked I don‘t see any with that „bad allocation“ error message.
I didn‘t found anything in the logs after the error appeared.

dartrax, 2024-03-31:
I haven‘t seen this issue with recent versions (v1.9.x) yet, I‘ll report back when it occurs again.

closed 2024-07-07 by github-actions[bot]