OrcaSlicer/OrcaSlicer #1921: State bed type in the "Send Print Job" window

FlyingLemming, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
As it is possible to select a different bed type for each plate, the plate to be printed is not always of the same bed type as displayed in the main GUI unter "Printer – Bed type". Hence, it happens every once in a while that the slicing has been performed for the wrong bed type. Usually this is covered by the printer – which still is a PITA as the build plate detection by the BL X1C takes place only after a couple of minutes have been wasted for other print preparations. Worse, occasionally the reading of the build plate code fails (e.g. when covered with some traces of glue) and one is presented with a message to be either confirmed blindly or resulting in an abort. In such cases, when the selected bed type is not the one intended and loaded into the printer, the print may fail or be of low quality.
In essence: it would be great to have a last-minute check of the _actual_ bed type before sending the file to the printer.

**Describe the solution you'd like**
I suggest to state the bed type for which the print has been sliced in the "Send print job to" or "Send to Printer MicroSD card" dialogues, e.g. between  the print time / filament mass and the AMS filament selection. The selected nozzle size would be another nice addition.

[image]

Noisyfox, 2023-08-26:
Plate types affect the generated gcodes I believe. Not sure how we could change plate type without re-slicing.

closed 2024-03-11 by github-actions[bot]

vanowm, 2024-12-07:
+1

_Apparently, ignoring the issue fixes it...what a magical time we are living in.
This bot is ridiculous._