OrcaSlicer/OrcaSlicer #1925: Crash if delete cube from pattern method PA, and slice. Doesnt show correct thumbnail either.

BakeJake, 2023-08-25:
**Describe the bug**
Crash if delete cube from pattern method PA, and slice. Doesnt show correct thumbnail either.

**3mf File for This Bug**

**To Reproduce**
Steps to reproduce the behavior
Goto calibrations from prepare mode
select pressure advance, select pattern
delete unwanted cube
slice
crash

**Expected behavior**
delete cube, and slice, and still have pa pattern.

**Screenshots**
[image]

**Printer model**
Ender 3 S1, Not Applicable though.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

Arakon, 2023-09-10:
Same behavior with 1.7.0-beta.

bani6809, 2023-11-08:
same behavior with 1.8.0-beta2

BakeJake, 2023-11-08:
@bani6809 It also doesn't output the exclude object right for that test, It meshes my whole bed when I have adaptive.

BakeJake, 2023-12-27:
> Dont delete that little cube - it's used as a handle to move the pattern around. Also best to disable exclude object when running this test.
> 
> Slim chance this will be corrected as its cosmetic issue.

Yeah thats bogus, it should be done internally we shouldnt see it placing that in the first place it should just work and do the pattern. should also display the pattern as an object after rendered so that it goes into the thumbnail. thats just lazy what you said lol.

igiannakas, 2023-12-27:
> Yeah thats bogus, it should be done internally we shouldnt see it placing that in the first place it should just work and do the pattern. should also display the pattern as an object after rendered [...]

The pattern is implemented as a custom gcode. You can see that in the vertical slider. Hence why it needs an object to tie it to. The object can be removed but then you won't be able to move the pattern around. It's not an STL so you can't see it in the prepare tab.

BakeJake, 2023-12-27:
> The pattern is implemented as a custom gcode. You can see that in the vertical slider. Hence why it needs an object to tie it to. The object can be removed but then you won't be able to move the [...]

You dont think it should be done internally? it should render an stl like OpenSCAD would. that would be ideal if youre worried about movement. No other slicer uses a placeholder for tuning that im aware of.

igiannakas, 2023-12-27:
> You dont think it should be done internally? it should render an stl like OpenSCAD would. that would be ideal if youre worried about movement. No other slicer uses a placeholder for tuning that im [...]

no other slicer implements this pattern. It's not an object, it's a custom gcode. Like the line pattern. The alternative is to display a text like the line pattern that is replaced when slicing.

closed 2024-04-03 by github-actions[bot]