OrcaSlicer/OrcaSlicer #1932: App crashes on linux on startup

vickas777, 2023-08-26:
**Describe the bug**
After update to 1.6.6 slicer became unusable on linux. On startup it freezes and after about minute closes itself. There are 2 different errors - on clean install and after network plugin installed.
On clean install, after network plugin instalation stacktrace:
```
systemd-coredump[20076]: [🡕] Process 18820 (orcaslicer_main) of user 1000 dumped core.
                                               
Stack trace of thread 20069:
#0  0x00007f41b889d3be free (libc.so.6 + 0x9d3be)
#1  0x00007f41aaf71c90 n/a (libcrypto.so.3 + 0x171c90)
#2  0x00007f41aafbff9b n/a (libcrypto.so.3 + 0x1bff9b)
#3  0x00007f41aafc3b27 n/a (libcrypto.so.3 + 0x1c3b27)
#4  0x00007f41b8889ac1 n/a (libc.so.6 + 0x89ac1)
#5  0x00007f41b888c860 n/a (libc.so.6 + 0x8c860)
#6  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18834:
#0  0x00007f41b890ee2d syscall (libc.so.6 + 0x10ee2d)
#1  0x00007f41be501ca7 g_cond_wait (libglib-2.0.so.0 + 0xafca7)
#2  0x00007f41be477144 n/a (libglib-2.0.so.0 + 0x25144)
#3  0x00007f41be4e12fe n/a (libglib-2.0.so.0 + 0x8f2fe)
#4  0x00007f41be4ded75 n/a (libglib-2.0.so.0 + 0x8cd75)
#5  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#6  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18835:
#0  0x00007f41b890365f __poll (libc.so.6 + 0x10365f)
#1  0x00007f41be509c2f n/a (libglib-2.0.so.0 + 0xb7c2f)
#2  0x00007f41be4aa0e2 g_main_context_iteration (libglib-2.0.so.0 + 0x580e2)
#3  0x00007f41be4aa132 n/a (libglib-2.0.so.0 + 0x58132)
#4  0x00007f41be4ded75 n/a (libglib-2.0.so.0 + 0x8cd75)
#5  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#6  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18869:
#0  0x00007f41b890365f __poll (libc.so.6 + 0x10365f)
#1  0x00007f41be509c2f n/a (libglib-2.0.so.0 + 0xb7c2f)
#2  0x00007f41be4abfef g_main_loop_run (libglib-2.0.so.0 + 0x59fef)
#3  0x00007f41b9d11f72 _ZN3WTF7RunLoop3runEv (libjavascriptcoregtk-4.0.so.18 + 0x1111f72)
#4  0x00007f41b9d13c5d n/a (libjavascriptcoregtk-4.0.so.18 + 0x1113c5d)
#5  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#6  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18887:
#0  0x00007f41b890ee2d syscall (libc.so.6 + 0x10ee2d)
#1  0x00007f41be502533 g_cond_wait_until (libglib-2.0.so.0 + 0xb0533)
#2  0x00007f41be477115 n/a (libglib-2.0.so.0 + 0x25115)
#3  0x00007f41be477277 g_async_queue_timeout_pop (libglib-2.0.so.0 + 0x25277)
#4  0x00007f41be4e1bee n/a (libglib-2.0.so.0 + 0x8fbee)
#5  0x00007f41be4ded75 n/a (libglib-2.0.so.0 + 0x8cd75)
#6  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#7  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18916:
#0  0x00007f41b890ee2d syscall (libc.so.6 + 0x10ee2d)
#1  0x000055f959560d3b n/a (orca-slicer + 0x339ed3b)
#2  0x000055f95954f019 n/a (orca-slicer + 0x338d019)
#3  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#4  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18917:
#0  0x00007f41b890ee2d syscall (libc.so.6 + 0x10ee2d)
#1  0x000055f959560d3b n/a (orca-slicer + 0x339ed3b)
#2  0x000055f95954f019 n/a (orca-slicer + 0x338d019)
#3  0x00007f41b888c9eb n/a (libc.so.6 + 0x8c9eb)
#4  0x00007f41b8910ebc n/a (libc.so.6 + 0x110ebc)

Stack trace of thread 18866:
#0  0x00007f41b88894ae n/a (libc.so.6 + 0x894ae)
#1  0x00007f41b888c055 pthread_cond_timedwait (libc.so.6 + 0x8c055)
#2  0x00007f41b9d78c54 n/a (libjavascriptcoregtk-4.0.so.18 + 0x1178c54)
[... 132 lines cut ...]
Stack trace of thread 22199:
#0  0x00007fad81ad58f5 clock_nanosleep (libc.so.6 + 0xd58f5)
#1  0x00007fad81ae7fb7 __nanosleep (libc.so.6 + 0xe7fb7)
#2  0x00007fad81b10a39 usleep (libc.so.6 + 0x110a39)
#3  0x00007fad58727ea1 n/a (/home/vickas777/.config/OrcaSlicer/plugins/libBambuSource.so + 0x127ea1)
ELF object binary architecture: AMD x86-64
```

**3mf File for This Bug**
None

**To Reproduce**
Start slicer, update network plugin

**Expected behavior**
App does not crashes on startup

**Screenshots**
None

**Printer model**

**Desktop (please complete the following information):**
 - OS: Manjaro Linux
 -  Kernel 6.4.9

tve, 2023-08-29:
I'm seeing the same, also on Manjaro.

outpoints, 2023-08-31:
Has there been any developments here?

R4toLoko, 2023-09-01:
I think this is an Arch problem, Mint Cinnamon ( DEB ) with an annoying javascript pop-up error pulled from BambuStudio but it works just fine.

outpoints, 2023-09-14:
Still crashes on official release.
This is the debug log

```
2023-09-14 12:27:14.167011[Thread 0x00007f3693cff600]:load_language: language , initial: 1
2023-09-14 12:27:14.167166[Thread 0x00007f3693cff600]:language provided by BambuStudio.conf: en_US
2023-09-14 12:27:14.174735[Thread 0x00007f3693cff600]:Update_dark_mode_flag: switch the current dark mode status to 1 
2023-09-14 12:27:14.176419[Thread 0x00007f3693cff600]:begin to show the splash screen...
2023-09-14 12:27:14.190108[Thread 0x00007f3693cff600]:loading systen presets...
2023-09-14 12:27:14.192665[Thread 0x00007f3693cff600]:[BBL Updater]:Checking whether the profile from resource is newer
2023-09-14 12:27:14.194557[Thread 0x00007f3693cff600]:Installing 2 bundles from resources ...
2023-09-14 12:27:14.202981[Thread 0x00007f3693cff600]:[BBL Updater]:Performing 8 updates
2023-09-14 12:27:14.202993[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL.json -> /home/user/.config/OrcaSlicer/system/BBL.json)
2023-09-14 12:27:14.204852[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/process -> /home/user/.config/OrcaSlicer/system/BBL/process)
2023-09-14 12:27:14.204857[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/process" -> "/home/user/.config/OrcaSlicer/system/BBL/process"
2023-09-14 12:27:14.238094[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/filament -> /home/user/.config/OrcaSlicer/system/BBL/filament)
2023-09-14 12:27:14.238108[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/filament" -> "/home/user/.config/OrcaSlicer/system/BBL/filament"
2023-09-14 12:27:14.312766[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/filament/P1P" -> "/home/user/.config/OrcaSlicer/system/BBL/filament/P1P"
2023-09-14 12:27:14.362773[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/machine -> /home/user/.config/OrcaSlicer/system/BBL/machine)
2023-09-14 12:27:14.362788[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/BBL/machine" -> "/home/user/.config/OrcaSlicer/system/BBL/machine"
2023-09-14 12:27:14.375051[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom.json -> /home/user/.config/OrcaSlicer/system/Custom.json)
2023-09-14 12:27:14.375872[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/process -> /home/user/.config/OrcaSlicer/system/Custom/process)
2023-09-14 12:27:14.375881[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/process" -> "/home/user/.config/OrcaSlicer/system/Custom/process"
2023-09-14 12:27:14.386261[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/filament -> /home/user/.config/OrcaSlicer/system/Custom/filament)
2023-09-14 12:27:14.386274[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/filament" -> "/home/user/.config/OrcaSlicer/system/Custom/filament"
2023-09-14 12:27:14.396437[Thread 0x00007f3693cff600]:	Update(/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/machine -> /home/user/.config/OrcaSlicer/system/Custom/machine)
2023-09-14 12:27:14.396451[Thread 0x00007f3693cff600]:copy_directory_recursively "/tmp/.mount_OrcaSl2PR5n8/resources/profiles/Custom/machine" -> "/home/user/.config/OrcaSlicer/system/Custom/machine"
2023-09-14 12:27:14.413822[Thread 0x00007f3693cff600]:initialize_network_module, successfully loaded library /home/user/.config/OrcaSlicer/plugins/libbambu_networking.so, module 0x55b895087a00
2023-09-14 12:27:14.413877[Thread 0x00007f3693cff600]:get_network_function, can not find function bambu_network_get_profile_3mf
2023-09-14 12:27:14.413883[Thread 0x00007f3693cff600]:on_init_network: on_init_network, load dll ok
2023-09-14 12:27:14.413885[Thread 0x00007f3693cff600]:get_network_agent_version=01.07.04.02
2023-09-14 12:27:14.413887[Thread 0x00007f3693cff600]:on_init_network: on_init_network, compatibility version
2023-09-14 12:27:14.421591[Thread 0x00007f3693cff600]:on_init_network, create network agent...
2023-09-14 12:27:14.422380[Thread 0x00007f3693cff600]:NetworkAgent, this 0x55b8950aaee0, network_agent=0x55b8950aaec0, create_agent_ptr=1
2023-09-14 12:27:14.429095[Thread 0x00007f3693cff600]:init_networking_callbacks: enter, m_agent=0x55b8950aaee0
2023-09-14 12:27:14.429156[Thread 0x00007f3693cff600]:init_networking_callbacks: exit, m_agent=0x55b8950aaee0
2023-09-14 12:27:14.430533[Thread 0x00007f3693cff600]:load_presets enter, substitution_rule 3, preferred printer_model_id 
2023-09-14 12:27:14.430658[Thread 0x00007f3693cff600]:load_vendor_configs_from_json: parse /home/user/.config/OrcaSlicer/system/Custom.json, got description:  "My configurations"
2023-09-14 12:27:14.436712[Thread 0x00007f3693cff600]:load_vendor_configs_from_json: parse /home/user/.config/OrcaSlicer/system/BBL.json, got description:  "the initial version of BBL configurations"
2023-09-14 12:27:14.473163[Thread 0x00007f3693cff600]:update_compatibility for all presets enter
2023-09-14 12:27:14.473186[Thread 0x00007f3693cff600]:update_compatibility for all presets exit
2023-09-14 12:27:14.473188[Thread 0x00007f3693cff600]:update_compatibility for all presets enter
[... cut ...]
2023-09-14 12:29:38.178002[Thread 0x00007f3693cff600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:4}
2023-09-14 12:29:38.178004[Thread 0x00007f3693cff600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:5}
2023-09-14 12:29:38.178005[Thread 0x00007f3693cff600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:6}
2023-09-14 12:29:38.178007[Thread 0x00007f3693cff600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:7}
2023-09-14 12:29:38.178015[Thread 0x00007f3693cff600]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:8}```

UndeadKernel, 2023-09-16:
The latest 1.7 release still has this issue (under Arch GNU/Linux).

I've noticed that this issue also happens with the official BambuStudio AppImage for Ubuntu but does not happen with the Fedora version.

vickas777, 2023-09-21:
I have mentioned that if you are not log in slicer works normally. Immediately after log in to bambulab profile it starts to crash.

simonvizzini, 2023-09-21:
Same issue on Arch Linux: when logging into my Bambu account slicer will crash on startup. I want to use the BBL account to easily share my filament profiles and printer settings across multiple machines. Is there maybe an alternative to easily share these settings?

pellcorp, 2023-11-08:
I was able to start 1.8.0 beta2 the slicer on Manjaro, I had issues running the 1.7.0 release on Ubuntu 23.10 (upgraded from 22.04 -> 23.04 -> 23.10), I will try the beta on Ubuntu on the weekend.

vickas777, 2023-11-19:
The issue has gone with stable 1.8 release.

closed 2023-11-19 by vickas777