OrcaSlicer/OrcaSlicer #1933: P1P .4 Nozzle profile keeps appearing

ZerkerEOD, 2023-08-26:
**Describe the bug**
I do not have the P1P selected as a printer I own under print settings but for some reason every time I open the print settings or load a new project it defaults to the P1P .4 nozzle profile causing print issues.

**3mf File for This Bug**
All 3mf's I tested.

**To Reproduce**
Start up the installed version of Orca Slicer on Windows
Start new project
Look at printer nozzle profile and it has the P1P selected even though the P1P is not a printer I own

**Expected behavior**
Only populate printers and nozzle profiles for the ones selected under the printer options.

**Screenshots**
[image]
[image]

**Printer model**
Bambu Lab X1C

**Desktop (please complete the following information):**
 - OS: Windows 11
 - App Version: 1.6.6

shaneo88, 2023-08-26:
I have a sneaking suspicion that the P1P nozzle is being loaded with the 3mf files you're loading. 

I loaded a 3mf file up today and instead of keeping my filaments/nozzles/settings it removed them all and loaded in all of these from the 3mf file. I am taking a guess that these settings are all from the person that made the 3mf.

ZerkerEOD, 2023-08-26:
@shaneo88, now I feel dumb. I was trying to print he Head Amame headset and normally when I open Orca Slicer it tells me I have an unsaved project and I just resume and delete the objects and throw a new file on. Looks like you are right, I kept printing other stuff and the profile was the same but it only loads when I load up one of the 3mf files from them. Thanks, I am still kinda new to this and this was driving me crazy.

closed 2023-08-26 by ZerkerEOD