OrcaSlicer/OrcaSlicer #1945: input_filename_base is inconsistent

nordurljosahvida, 2023-08-28:
### Bambu Studio Version

1.6.6

### Where is the application from?

Bambu Lab github releases

### OS version

win11

### Printer

x1 carbon

### How to reproduce

- open a new project
- select any bambu printer [i'm using x1 carbon]
- import any geometry or test cube
- save the project with a custom name
- press "slice"
- press "export gcode file"

or with any custom printer:

- open a new project
- select a non-bambu printer [any custom one you created, i'm using one based on the generic klipper printer that can be added on first setup of orca]
- add any geometry or test cube
- save the project
- press "slice"
- export gcode or press the "print" button

### Actual results

the name of the file will be the name of the object being printed, not of the actual saved project name

### Expected results

The exported gcode name includes {input_filename_base} as per output options setting. However, this variable should the project name, not the imported model geometry file's [the stl as an example]. In prusaslicer, it's exactly like this. Even in bambu and orca, when sending a file to a bambu printer, the name is of the projects, this is only an issue in how input_filename_base is processed.

Of course, if a project is saved, then the exported gcode - whether exported locally or to sd card or sent over directly to printer via octoprint / klipper - should match that name. I think this problem was introduced in bambu studio, but is very easily fixable.

Thanks!

### Project file & Debug log uploads

no file as this should be tested from scratch with any geometry

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SylvainKlein, 2023-11-13:
Having the exact same issue here. Would love a fix for that ! :)

Matt2916, 2023-12-05:
Still present in Orca 1.8.1 - would love a fix for this!

UnseenStyle, 2024-03-03:
Would love the ability to label bambu print filenames to the plate name instead of it being just the project name.

vale075, 2024-05-26:
> Would love the ability to label bambu print filenames to the plate name instead of it being just the project name.

It is already possible by using: plate_name

jbjhjm, 2024-07-16:
A fix for this would be much appreciated. Currently I need to manually change the file name for each export.

jbjhjm, 2024-08-14:
I took a moment to check the source code. 
Interestingly, the implementation has inconsistent concepts in all code bases, be it orca, prusa, superslicer or bambu.

https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/PrintBase.cpp#L58
contains two functions defining different values to input_filename / input_filename_base.

In PrintBase::update_object_placeholders these two are set to the INPUT MODEL filename.
In PrintBase::output_filename these two are set to the PROJECT file name.

The latter one will only everwrite the config values though if a non-empty filename_base argument is being passed into the function. I'm currently assuming that for some reason this is not happening for Orca anymore.

It seems that the output_filename function is regularly called by Print::output_filename.
That's where I'm stuck right now, I cannot seem to figure out what's calling that function and passing a valid/invalid file name.

The existing implementation seems questionworthy, when "fixing the override" I guess other users may be irritated.

jbjhjm, 2024-11-13:
chu bot, nothing stale here!

Jeppedy, 2024-11-13:
Yup, just trying out OrcaSlicer, and that's the first thing I noticed..  My filename has "body1" as the name, and not the ACTUAL project file name.  I could use plate_name, but I still have no reliable way to introduce the "project" name to the filename.

ShryuKt, 2024-11-19:
I'm looking for the same thing

poulanker, 2024-11-21:
I also noticed, that no matter which AMS spool number you select for slicing, the filament type inserted in the gcode filename always follows whatever filament type is selected for spool #1 .

Jeppedy, 2024-11-21:
For those looking for a quick hack/fix, I set my model name in the Project tab and use the following Filename template:
{if model_name!=""}{model_name}_{endif}{if plate_name!=""}{plate_name}_{endif}{if filament_type[0]!=""}{filament_vendor[0]}{filament_type[0]}{else}NoFilamentType{endif}_{nozzle_diameter[0]}.gcode

blalor, 2025-02-20:
Not stale, still exists. The current behavior is really weird, especially coming from PrusaSlicer.

golovashevich, 2025-05-22:
Still exists

greencheetah100, 2025-11-12:
I still have this issue in 2.3.1

{input_filename} and {input_filename_base} both return Object_1, the default first object name. Same result with custom printer and ender 3 preset.

closed 2026-05-20 by SoftFever