OrcaSlicer/OrcaSlicer #1946: custom or personal process setting lost when Poste-Processing Script is present

adop-adop, 2023-08-28:
### Bambu Studio Version

Orcaslicer 1.6.6.

### Where is the application from?

Bambu Lab github releases

### OS version

Win 10 X64

### Printer

Custom Idex Printer based on TL-D3S

### How to reproduce

Open Orcaslicer 
add model on bed
modify the Process setting, Speed, Wall Loop, other, ....
and importante step : add a Poste-Processing Script in : Process/Other/Poste-Processing Script
save this setting as "User Preset" put name you like (exp: My best setting)
slice
print ... ok

CLOSE the Orcaslicer
and reopen him 

the Process settings that you just create and save is not present.... all your setting "User Preset" named "My best setting" in my example are lost

But If you not write a script in Process/Other/Poste-Processing Script, the the User Preset is saved and present for the next Orcaslicer opening

The problem is I can't save my Process User Preset with Script for Poste-Processing 

### Actual results

the User Preset are lost when you write in  Process/Other/Poste-Processing Script.

### Expected results

I would like to save my User Preset with the Poste-Processing Script.

Thanks.

### Project file & Debug log uploads

Any project file need to reproduce the issue

### Checklist of files to include

- [ ] Log file
- [ ] Project file

foreachthing, 2023-08-28:
Could it be the same as here: https://github.com/SoftFever/OrcaSlicer/issues/1397  or here https://github.com/SoftFever/OrcaSlicer/issues/1435#issuecomment-1616434951 ?

adop-adop, 2023-08-31:
> 

Yes It's same issue

PrusaSlicer team please take some time on this problem, it's very uncomfortable to list the User preset setting

Thanks.

closed 2023-12-08 by github-actions[bot]