OrcaSlicer/OrcaSlicer #1954: Prusaslicer 3MF format support

arshish1612, 2023-08-29:
Is there any plan to support Prusaslicer 3mf files?

Vanilla009, 2023-08-29:
I would also like this-I have some files with custom supports I'd like to be able to use. Also the 3mf import option from prusaslicer.

arshish1612, 2023-10-27:
Bump Bump

arshish1612, 2024-01-29:
Pls keep this active

mlewis-everley, 2024-02-06:
I would really appreciate re-adding support for Slic3r based 3MF. I am slowly migrating about 100 3MF files from SuperSlicer to Orca and each one has one (or all) of the below:

1. Custom height range modifiers
2. Negative parts
3. Custom geometry modifiers
4. Custom object settings

None of the above are supported when importing a 3MF.

As a bare minimum, it would be great if the importing a Slic3r 3MF could handle more complex assemblies (extra parts, negative parts or modifiers), even if it doesn't import custom print settings.

Even better, it would be great if the actual print settings were supported in some way.

closed 2024-05-14 by github-actions[bot]

fgeck, 2024-05-22:
I would really love to see this feature. I want to print a lot of projects from printables.com and most people share 3mf files