OrcaSlicer/OrcaSlicer #1958: Ability to add Printer by IP address (Lan Only Mode) for Bambu Lab printers

fire1ce, 2023-08-30:
Lan only mode uses auto-discovery to find a printer on local network.
When using remote connection to the lan segment - like VPN, auto-discovery not always works and there is no way to use the printer (Communication to the mqtt works fine).

Add the ability to add printer manually hard coded by ip address / hostname (same as for klipper)

odious, 2023-12-02:
+1 this would be an amazing addition. 
at the moment, the discovery of printers on the local network has to be faked by generating those SSDP pakets yourself (there's a python script for this out there), which is a big hassle and not at all feasible for anyone.

also, auto connect to a specific lan mode printer would be great, maybe even just remember the last one connected.

fpetzold, 2024-01-09:
I would also love to see this feature. I need it to connect to the printer via VPN.

rehashedsalt, 2024-02-09:
+1. Bambu upstream has expressed little interest in adding this feature thus far.

toscdesign, 2024-05-05:
Would also like to have this feature!
This would be a big benefit for the Orca Slicer!

tobiasium, 2024-05-06:
+1 would be great to have this feature!

talentedmrripley, 2024-05-17:
@odious Can you share any info on the script?

odious, 2024-05-17:
> @odious Can you share any info on the script?

I stumbled upon this one: https://github.com/jonans/bsnotify
Haven't checked if it works though. 

An option to auto-connect the slicer to a configurable printer would be a lot less hassle than having to fake the discovery pakets.

talentedmrripley, 2024-05-17:
I found and used this one successfully yesterday - https://gist.github.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3

HanBrol0, 2024-06-29:
+ 1

Triskae, 2024-07-17:
Is there any news about the progress of this issue ? Is there something purely on the slicer side, or is the slicer community waiting for bambu to change something in the FW of the printers ?

tinuva, 2024-10-19:
Also interested to see this added to Orca slicer.

https://github.com/bambulab/BambuStudio/issues/702#issuecomment-2424049503
Mentions bambu studio added this to studio 1.10 and the public beta release is available, with the feature added and available.

When can or should we expect similar feature availability in orca slicer?

Alex-Schaefer, 2024-10-19:
This is the relevant commit with the changes to add the new binding method: https://github.com/bambulab/BambuStudio/commit/0ec49c3588c1f81c159193de352f8a47d39ffe74

I had no time to get the build process of OrcaSlicer properly running, so I can't test and implement this, but I would love to see this implemented in OrcaSlicer, too.

tinuva, 2024-10-20:
ah damn, looks like the feature is not exactly working yet, others also reporting it broken https://github.com/bambulab/BambuStudio/issues/5006

rocket59, 2024-11-15:
Looks like Bambu will also be releasing new printer firmware as well because of these issues?

Also I look forward to this finding its way to Orca soon!

emeraldtip, 2025-01-19:
Especially with the recent announcements from bambu, this is still needed as I have the printer locked away on a separate subnet that can't communicate with the outer world

Duglum, 2025-01-19:
Easiest would probably to just implement a functionality like this one: https://gist.github.com/fritzw/7d981046b509a27e204d01c9bd73f37b

So, just sending out a proper discovery packet. I'm running this script via task manager and it works fine.
Since Bambu has just shown that we will not be able to install any new firmware going forward without losing functionality this is probably the only way.

probonopd, 2025-01-19:
Strange thing is that it used to work in the past...

lukasmalkmus, 2025-01-20:
Just ran into this problem. My X1C is logged into my IoT WiFi which is on a different subnet.

emeraldtip, 2025-01-20:
I can vouch for the gist by fritzw above, that works fine for connecting to things on a different subnet

DissBee, 2025-01-20:
+1 workarounds are fine but would be excellent to have this be added - especially given the "security changes"

tinuva, 2025-01-21:
Yeah I don't understand why Orca can't just add a piece of code that generate these SSDP packets for the plugin when added manually by ip address. That said, the plugin is falling away as I see it so a more generic way would be, ability to add printer by ip address, have the new advanced developer mode enabled, so all orca have to do, is upload the model by sftp and then initiate the print (if we have this option).

talentedmrripley, 2025-01-21:
> Easiest would probably to just implement a functionality like this one: https://gist.github.com/fritzw/7d981046b509a27e204d01c9bd73f37b
>
> So, just sending out a proper discovery packet. I'm running [...]

What is the purpose of running the script as a scheduled task?

rocket59, 2025-01-22:
> Easiest would probably to just implement a functionality like this one: https://gist.github.com/fritzw/7d981046b509a27e204d01c9bd73f37b
> 

+1, yes please!

SgtKilgore406, 2025-01-25:
+1 Please implement this functionality. My X1C lives on an IoT VLAN and recently had LAN Only mode enabled and was blocked from accessing the internet in wake of Bambu Lab's announcement.

Uninstalled Bambu Studio and installed OrcaSlicer only to find out that I cannot add my printer to OrcaSlicer due to it being on a different subnet.

mxlje, 2025-01-26:
This would be super nice, especially following the recent controversy and many more people switching to LAN mode. Also for tools like tailscale and generally more complex network setups, entering the IP directly would be crucial. Thank you.

rwi11mann, 2025-01-26:
> +1 Please implement this functionality. My X1C lives on an IoT VLAN and recently had LAN Only mode enabled and was blocked from accessing the internet in wake of Bambu Lab's announcement. [...]

Same situation here with a P1S.

VegaMyssil, 2025-01-29:
+1 I have 3 printers unable to reach on other subnet ...

DerSchinken, 2025-02-09:
+1 My printer, which is in lan only mode, sometimes (it's almost all times actually) doesn't appear with the auto-discovery on my laptop even tho it's on the same network and I can even ping it (my guess is that it's because of the printer being connected to a different repeater than my laptop sometimes, but idk).

rocket59, 2025-02-10:
Have tried this yet? Working fine for me.

[](https://github.com/SoftFever/OrcaSlicer/pull/8256)

fanaticDavid, 2025-03-01:
Can confirm that the latest beta of OrcaSlicer works for me. Yesterday, I decided to set my X1C to LAN Only mode, only to find out that BambuStudio wouldn't connect to my printer anymore. My X1C is on a separate IoT VLAN, while my desktop computer and laptop are on my main VLAN.

These findings apply to both Windows and macOS, despite the note about macOS in #8256.

tinuva, 2025-03-02:
Oh woah what an improvement with the latest 2.3.0 beta2. Can confirm, no longer need the script to be able to use printer.

lin-ycv, 2025-03-02:
Anyone using the lastest beta can confirm whether orca still randomly forgets the pass code when using multi printer setup in lan?

closed 2025-06-09 by github-actions[bot]