OrcaSlicer/OrcaSlicer #1964: Top layer with terrible inconsistency

R4toLoko, 2023-08-31:
### Bambu Studio Version

1.6.6

### Where is the application from?

Bambu Lab github releases

### OS version

Mint Cinnamon 21.1

### Printer

LDO Voron Trident 300mm

### How to reproduce

1. Slice a model

### Actual results

Top layer is creating this terrible inconsistency which not just increase the printing time but also terrible top layer finish.
No matter the wall generator you pick, you have this ugly finish.
I mean, look at this HDD.

[image]

[image]

### Expected results

PrusaSlicer on the right is the expected result with no weird top layer.
This is new and have been introduced recently!!

[image]

**IMPORTANT:** Please, fix this bug report, it looks a copy/paste from BambuStudio instead of OrcaSlicer.
Linux log for example is located under `$HOME/.config/OrcaSlicer/log`
Noobs will have a hard time following this.

Thank you

### Project file & Debug log uploads

[3MF.zip](https://github.com/SoftFever/OrcaSlicer/files/12483983/3MF.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

R4toLoko, 2023-08-31:
This happens across different models so this is not a model, it happens with the project attached and any other project.

R4toLoko, 2023-08-31:
[debug_Thu_Aug_31_18_15_35_36459.log](https://github.com/SoftFever/OrcaSlicer/files/12484070/debug_Thu_Aug_31_18_15_35_36459.log)

R4toLoko, 2023-08-31:
Also, PrusaSlicer approach with the hook part makes more sense, it will also prints faster and better top layer quality.
On OrcaSlicer to do so, you need to disable the top surface scheme.

[image]

The hook part only, it makes more sense than this.
I wonder if this isn't all related.

Thanks

[image]

Noisyfox, 2023-08-31:
Have you tried disabling the "single wall on top surface" option?

R4toLoko, 2023-08-31:
> Have you tried disabling the "single wall on top surface" option?

Are you joking with me?? lol
I have single wall on top surface for awhile now but only now I notice the behavior above and following your recommendation solved the problem.

Thank you 
[image]

closed 2023-08-31 by R4toLoko