OrcaSlicer/OrcaSlicer #1968: Extra perimeter on overhangs results in missing bridges / additional supports

jesterhead82, 2023-08-31:
### Bambu Studio Version

1.7.0-dev

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Printer

Selfmade

### How to reproduce

Without Extra Perimeters for Overhangs the slots are properly bridged:
[image]

With Extra Perimeters for Overhang some bridges get replaced by Overhangs:
[image]

Enabling "Dont support bridges" will print those long overhangs instead of short bridges.

### Actual results

Overhangs instead of bridges

### Expected results

Bridges instead of Overhangs

### Project file & Debug log uploads

[Gen 2 Lack Case - Double.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12490161/Gen.2.Lack.Case.-.Double.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tome9111991, 2023-09-03:
in my opinion no bug it is exactly what Extra perimeter on overhangs does.

jesterhead82, 2023-09-04:
At least it should be either everywhere the same (because it generates only in some areas the extra perimeters).

Also SS does generate generate Bridge Infill type correctly for that Usecase properly with Extra Perimeters for Overhangs:

[image]

tome9111991, 2023-09-04:
the feature is ported from PS. maybe SS has a different behavior.
maybe @Noisyfox  can say something about it

closed 2023-12-11 by github-actions[bot]