OrcaSlicer/OrcaSlicer #1969: FR: Time estimate calibration

Arakon, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
The print time calculation is off by about 25% or so in most cases on my VZBot running Klipper, despite accurate speed and acceleration values being entered in Orca. In SuperSlicer, it's possible to adjust the estimate by a percentage factor, which makes the already fairly accurate calculation (within 10%) even more accurate.

**Describe the solution you'd like**
It'd be great if the print time estimate could be reworked completely or at least a percentage modifier added.

kylek29, 2023-09-06:
I think the time estimate algorithm might just be bugged 
at the moment  (possibly a change BambuLabs did?) -- I did a print with X1C with standard BambuStudio, estimated at roughly 3 hours, took 4 hours.

It was always fairly accurate with Orca and BambuStudio before they BambuStudio 1.7 changes.

JorgeSDF, 2023-12-08:
I have the same problem with my VZBOT 235, I have check SQV and Jerk on the slicer and both are the same. Estimates are like 11h print on slicer and real time of 5 hours...
[image]

HakunMatat4, 2023-12-16:
Big prints are often shown 1h up to 4h-ish longer than Mainsail/Klipper shows.
I was avoiding some prints at the middle of the day because it would finish plus 0AM and I forgot to don't trust OrcaSlicer time.

@HumbleDeer as far as I am aware, you don't wanna play around jerk or things go sideways.
Setting those settings to zero sounds...... wrong.

AByzhynar, 2024-01-13:
Any progress with this issue? Any plans ?

HumbleDeer, 2024-01-13:
> @HumbleDeer as far as I am aware, you don't wanna play around jerk or things go sideways. Setting those settings to zero sounds...... wrong.

Setting it to 0 just disables setting of any values through gcode. As far as I'm aware, no gcode is emitted for setting jerk on the printer when 0 is set as value on OrcaSlicer.

> Any progress with this issue? Any plans ?

For any of the changes that were proposed in my pull request, I'd suggest contacting someone on the @vzbot3d team. I'm no longer with them, and don't have any contact with anyone regarding these changes. If they would still like them implemented, I can still submit them.

closed 2024-04-20 by github-actions[bot]