OrcaSlicer/OrcaSlicer #1970: Weird top layer

klmn-roland, 2023-09-01:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 64bit

### Printer

X1C 

### How to reproduce

Slice a model with thin text on its top surface

### Actual results

Bad quality prints

### Expected results

Top surface like in bambu studio

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-09-01:
Try disable the single wall on top surfaces option.

klmn-roland, 2023-09-01:
Yeah this worked, thanks. But bambu studio works fine even if it's turned on.

Noisyfox, 2023-09-01:
Yeah improving this is on the list.

closed 2023-09-01 by klmn-roland