OrcaSlicer/OrcaSlicer #1972: Part cooling Fan speed control settings

kot0005, 2023-09-02:
[image]

could we please get the settings for fan speed control, like superslicer has ?

I was going to try orcaslicer on my klipper printer but noticed it doesnt have any fine tuning controls for part cooling fan.

i have included a screenshot of the settings that superslicer has.

movinimage, 2023-09-02:
This!

It is sometimes critical to change the fan speed for Internal bridging AKA the first solid layer on top of infill...which Prusa derivatives see as a bridge...hard-to-print-warp-prone filaments like ASA need this for internal bridges that are large and close to the bed.

The first solid layer on top of infill is considered a bridge and therefore uses the bridging fan speed. This is fine for lot of filaments...but high-warp ones like ASA can't take an entire large layer of 80% fan...especially if it is only 1/4" off the build plate at that time....it consistently warps the part and all the corners come off the build plate, and the part is ruined.

The only way to fix it currently is to sit by the printer and when a large, low internal bridge starts to print, manually lower the fan.

jeremytodd1, 2023-09-03:
Agreed. These settings would be really handy to have in some cases.

I recently had an issue where a 2 layer thick part of an object had a shiny finish. I then realized that my part cooling fan is only set to turn on after layer 3, which is a good thing as that will help prevent the first layer from warping.

To fix the shiny surface on the 2nd layer though, I had to set the filament's cooling to start the fan after layer 1.

If I had these settings I would have been able to just set the "top layers fan speed" and it would have been perfect.

I'm all for these settings to be added.

kot0005, 2023-10-03:
Hi, any update on having these settings in Orca slicer ?

nfons, 2023-11-10:
+1 to this....when using larger layer heights, part cooling sometimes warps the prints since it doesnt allow cooling fast enough at heigher heights.

closed 2024-02-20 by github-actions[bot]

VladKolerts, 2024-02-28:
how to switch off fan?

philbert440, 2024-03-19:
can we reopen this?

Novichek-13, 2024-04-04:
who has figured out how to completely turn off the airflow when printing? It gets in the way when printing ABS

fantastic-engineering, 2024-06-16:
For anyone still looking for fan controls, they are not in the printer settings, they are in the **filament** settings.
[image]
[image]
Hope this helps :)

VladKolerts, 2024-06-18:
> For anyone still looking for fan controls, they are not in the printer settings, they are in the **filament** settings. 
We know where fan control settings)
Our question "**How to switch off the fan?**" 
You can just set  "No cooling for the first" to 99999 - not cool

Maxell2904, 2024-06-19:
This is a big issue for me. Please improve fan controls.

GrigoriyPeters, 2024-06-21:
It would be nice to have this issue reopened.
Inability to fully disable fan is frustrating.
And it would be cool, to disable fan cooling in project settings.
Like I know which parts is stronger without cooling and vice versa.

SarraceniaWilds, 2024-06-30:
[image]
Screenshot to illustrate request. Lack of feature specific control for fan, is there an open issue for this request? #5328? #4370?

Asiak3, 2024-09-01:
Any fix for this?

KubaMiszcz, 2025-03-20:
> For anyone still looking for fan controls, they are not in the printer settings, they are in the **filament** settings. [image] [image] Hope this helps :)

and how turn default fan to 80% for whole print? excepr exclusions,

KubaMiszcz, 2025-03-20:
> It would be nice to have this issue reopened. Inability to fully disable fan is frustrating. And it would be cool, to disable fan cooling in project settings. Like I know which parts is stronger without cooling and vice versa.

also it is good if i can set to 80% eg

Lensaic, 2025-08-29:
Would love a "fan blocker" modifier that could be added to certain portions of parts. Printing ABS-ASA multi-material part where the top layer of the material transition, while completely solid, is considered a bridge and gets the bridge fan speed. Because this is close to the bed it has been warping the part. Other internal bridges need the modified fan speed, and I haven't found a way to reduce the fan just for this part of the print.