OrcaSlicer/OrcaSlicer #1979: 1.7 Dev - Organic support slicing errors out when support interface material is selected.

igiannakas, 2023-09-02:
### Bambu Studio Version

1.7 Dev

### Where is the application from?

Built from source by yourself or others

### OS version

OSX

### Printer

X1  Carbon

### How to reproduce

1.7 Dev - Organic support slicing crashes when support interface material is selected.

Open attached 3mf file, Slice & it errors out
[image]
[3DBenchy - Organic.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12504829/3DBenchy.-.Organic.3mf.zip)

### Actual results

Slicing crashes

### Expected results

Slicing completes with organic support interface material being the interface filament.

### Project file & Debug log uploads

[3DBenchy - Organic.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12504830/3DBenchy.-.Organic.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-09-03:
This is not an issue of the organic support, rather an error in your filament change code. The error happens even with other types of supports.

SoftFever, 2023-09-03:
> This is not an issue of the organic support, rather an error in the filament change code. The error happens even with other types of supports, or with multi-color enabled.

That's right.
This is an invalid variable name error. Check your custom filament change gcode. Fix the variable name will solve your problem.

Noisyfox, 2023-09-03:
I noticed that this issue doesn't exist until commit 05194ab1f4c90ebcf8c5c13db2041dc26472fc14. I guess there could be some issue there.

SoftFever, 2023-09-03:
> I noticed that this issue doesn't exist until commit [05194ab](https://github.com/SoftFever/OrcaSlicer/commit/05194ab1f4c90ebcf8c5c13db2041dc26472fc14). I guess there could be some issue there.

Damn, it's indeed a regression.
Are you looking into this issue? If not I can take a look

Noisyfox, 2023-09-03:
I think the issue is caused by this line:
https://github.com/SoftFever/OrcaSlicer/commit/05194ab1f4c90ebcf8c5c13db2041dc26472fc14#diff-e1f449c8c0a38f6b29108346154f3c4dc8a608298eae806e5e534d87cfe1d3c0R887

igiannakas, 2023-09-03:
Wow thank you both!

igiannakas, 2023-09-03:
Pulling a local build to retest 👍🏻🙏🏻

igiannakas, 2023-09-03:
Working. Closing issue, thank you both!!

closed 2023-09-03 by igiannakas