OrcaSlicer/OrcaSlicer #1983: Automatic support painting

skullbone, 2023-09-03:
Any plan on porting automatic support painting?
I find that It's very fast and convent way of minimizing unneeded support and just giving a good base for further control over the supports if needed. So it would be great to get it on orca

DVSVIDEO, 2023-09-03:
Support painting already exists in Orca slicer.

skullbone, 2023-09-03:
They do, but they aren't the same thing. I find that automatic support painting provides the optimization and control of manual supports with the quick and convenience of automatic supports. Using those two things together can make something similar, but it won't be quick or convenient.

DVSVIDEO, 2023-09-07:
Maybe it's the language difference (English is not my mother tongue) but I don't understand you.
The slicers either have support turned off or on. If it is on, the slicers will automatically generate support for overhangs. This is called automatic support.
If you are not satisfied with this, you can manually add or disable the support. This is the manual method and this is the painting of the underlay.
How do you mix these two separate methods? Either automatic or painted (manual)

lewis2e, 2023-09-07:
@DVSVIDEO Automatic support painting is sort of half way between manual and automatic supports. It's automatic, but more sophisticated than just automatically enforcing supports based on overhang.

From the [Prusa support page](https://help.prusa3d.com/article/paint-on-supports_168584#automatic-painting):

_The algorithm that powers this tool inspects the print extrusion by extrusion, considering a multitude of factors such as the center of mass, weight of the supported part, bed movement, potential extruder collision, bed adhesion, material, and bridging. The result is a more efficient placement of supports. For instance, bridges are supported only by regularly spaced pillars instead of their entire length._

@skullbone is right. It is really cool, works well in PrusaSlicer. It would be great to have in Orcaslicer. Given that it exists upstream in PrusaSlicer, maybe it will make its way down eventually.

DVSVIDEO, 2023-09-12:
I still don't understand: if it's automatic, it's not painting... It is automatically placed by the slicer. It doesn't paint, it just creates it.
Manual placement is called painting, where I mark with a tool where I want the support. It's not automatic, it's manual.

How can you paint by hand and generate automatically at the same time?

csn495, 2023-09-21:
I print a lot of large models that unfortunately need a lot of support, I have found that using the automatic support painting feature in prusaslicer makes it so much easier / faster to use the minimal amount of support material possible while also supporting the models really well.

I came across this while preparing to create a feature request for the same thing. In my opinion it is a game changing feature and it would be really awesome if it came to Orcaslicer.

Savij, 2023-11-16:
In Prusa slicer you can use paint on supports, in that dialog there is a button for automatic painting. Clicking the automatic painting button is similar to turning on supports, but it is more accurate. The biggest reason to use that feature IMO is that once the automatic supports are painted onto the model, you can then add your own support paint/blockers and add to, or remove the automatic painting that was done. In Orca slicer I have to choose either manually painting all supports, or using the automatic support. I can't do both (at least I have not figured out how). How can I turn on automatic supports and then remove just a section of those supports? I can do that with prusa slicer. I think that's what the OP is looking to do here and I miss that feature too. I actually switch over to prusa slicer when I need to do it, but I prefer to use only Orca slicer when i can.

gyordanov, 2023-11-20:
Also prusaslicer generates way better supports that way, less support material and easier to remove .. I'm not sure why that is not the default option for "auto supports"

closed 2024-02-27 by github-actions[bot]

libelred, 2024-10-22:
this is the only real feature i am missing from prusaslicer. i have models that i could print just like that with it, and are a continuous failure with the automatic ones

Aiddy81, 2025-05-19:
This is a feature that I use often and is quite handy it's strange that this as being marked as not planned for Aug leiser's I think it would be a great addition