OrcaSlicer/OrcaSlicer #1989: Gcode template broken for Flsun QQ-S Pro

deac05, 2023-09-03:
### OrcaSlicer Version

1.6.6

### OS version

Windows 11

### Printer

Flsun QQ-S Pro

### How to reproduce

Import Model and slice

### Actual results

Model does not slice and comes up with error 
Failed to generate G-code for invalid custom G-code.
machine_start_gcode Parsing error at line 51: Variable does not exist
M117 Print [output_filename_format]; Display: Printing started...
            ^
Please check the custom G-code or use the default custom G-code.

*This happens to any model i try

### Expected results

Model should slice and be ready to print

### Project file & Debug log uploads

[THRDHD8xTHR12.zip](https://github.com/SoftFever/OrcaSlicer/files/12505636/THRDHD8xTHR12.zip)
[debug_Sun_Sep_03_10_16_38_24044.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12505637/debug_Sun_Sep_03_10_16_38_24044.log.zip)
[debug_Sun_Sep_03_09_59_37_13140.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12505643/debug_Sun_Sep_03_09_59_37_13140.log.zip)
[Lens Cap Holder 58mm.zip](https://github.com/SoftFever/OrcaSlicer/files/12505644/Lens.Cap.Holder.58mm.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

foreachthing, 2023-09-03:
Looks like the variable `[output_filename_format]` does not exist or is misspelled.
Try this `[input_filename_base]` instead. This is just to display what model (filename) is printing. If you need it change it, if you don't remove it.

closed 2023-12-11 by github-actions[bot]

jwnremt, 2024-01-06:
I’m having this same issue.

ikaro-viana, 2024-06-26:
tambem tenho o mesmo problema, alguem conseguiu resolver?

vitaliyhan, 2025-05-03:
> Looks like the variable `[output_filename_format]` does not exist or is misspelled (in machine settings, start gcode). Try this `[input_filename_base]` instead. This is just to display what model (filename) is printing. If you need it change it, if you don't remove it.

that worked for me 

printer profile->G-code of printer -> star code -> search for the `[output_filename_format]` at the end change to `[input_filename_base]`