OrcaSlicer/OrcaSlicer #1990: Post processing script causes Orca to wipe user profile

Arakon, 2023-09-03:
### OrcaSlicer Version

1.6.6

### OS version

Windows 10

### Printer

VZBot custom

### How to reproduce

- Create a user profile
- Trying to run the following post processing script (note the quotes): 
**"K:\3D Printer\klipper_estimator.exe" --config_moonraker_url http://192.168.0.66 post-process**
- Close Orcaslicer
- Userprofile is immediately deleted now, or on next start

The post processing script runs and works, so it seems that using quotes causes Orca to assume the config is corrupted, and deletes it. Manually removing the post process line and copying the backup file back into the folder results in the preset being kept.

### Actual results

- Config is wiped out

### Expected results

Config disappearance and lots and lots of headache should not occur.

### Project file & Debug log uploads

[debug_Sun_Sep_03_13_24_49_3792.log](https://github.com/SoftFever/OrcaSlicer/files/12505829/debug_Sun_Sep_03_13_24_49_3792.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

foreachthing, 2023-09-03:
Dup https://github.com/SoftFever/OrcaSlicer/issues/1397

Arakon, 2023-09-03:
Indeed it is. Sorry, seems my search skills failed me there.

closed 2023-09-03 by Arakon

foreachthing, 2023-09-03:
You just need to make sure you don't need the quotes and you're fine. 😄 
Rename your directories, so they don't contain spaces.

Arakon, 2023-09-03:
Not an option for me since the folder contains all my slicers and 3d printing related stuff, so I'd have to reinstall or change paths for everything.
I did however find that you can simply dump the exe or scripts directly into the Orcaslicer folder and call them without any path at all.