OrcaSlicer/OrcaSlicer #1995: Create a local library of models

avxmw, 2023-09-03:
> **Is your feature request related to a problem? Please describe.**
> A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I have a every growing collection of models I want to (re-)print, many of which consist of multiple parts/build plates. This kind of data is hard to store accessible on a normal filesystem and to my knowledge there is no dedicated database around for this(at least on Linux). While sites like Thingiverse or Printables offer collections of some kind, they are plagued by outtages, bad search/filters and can't store print metadata or gcode for *your* printer.

> **Which printers will be beneficial to this feature**
> E.g. Voron/Klipper based printer/etc...
Pretty much every printer - in the sense of human not machine

> **Describe the solution you'd like**
> A clear and concise description of what you want to happen.
I'd like to see a library populated with my (downloaded) models which is closely tied into my slicer and which offers features like

* previews: screenshots or 3d rendered
* tags and folders
* store settings for a model, but not actual gcode. Define things like material, temps, positions, but re-slice on request to take advantage of the next-big-thing automatically (like Arachne was for example)
* when imported print-ready from the slicer import data like filament length needed automatically so it's easy to find stuff you can print with the last few loops on a roll
* link to custom shortcuts, for example open source file to edit in favorite CAD software

> **Describe alternatives you've considered**
> A clear and concise description of any alternative solutions or features you've considered.
Continue living like a peasant always getting Murphy-d by printables when you need that 1 file just now and spend hours sifting through obscurely named files without previews in the hope you might have downloaded it years before.

> **Additional context**
> Add any other context or screenshots about the feature request here.
I'm not sure if this is the best place to ask for something like this or if this should be stand alone software, but this is probably the best community to ask since Prusa is already going more and more into the direction of monetizing Printables.

kylek29, 2023-09-06:
I think the 3MF standard only stores a lot of that information, so are you basically looking for an in-program search (and indexed) of a model library that can search on that metadata (3MF by nature puts it into the .zip).

Like a folder the user would set as their "model library" and Orca would have a view to browse/search that library?

avxmw, 2023-09-07:
It doesn't need to be in-proram, but slicer and this should work closely together and exchange data where it makes sense - like Photoshop+Lightroom for example.

Just like we use media players to play "Rock" from the "70s" and rely on metadata to find it instead of hunting down the filesystem by hand I'd like to have a software doing that for 3d modeling files.

Once you've got 100+ models or your interests are very broad it is very complicated, labor intensive and needs extra hdd space if you try to do all this by hand with a common filesystem only.

"TheApp"/Orca would need to provide decent search with flexible tagging system (as in free form additionally to pre-defined) while the user does the tagging.

Again, doesn't need to have the "orca brand", but this is the crowd which a) most likely would use something like this and b) has the minds to actually create it.

closed 2023-12-15 by github-actions[bot]