OrcaSlicer/OrcaSlicer #2003: Implement Klipper Estimator by Annex-Engineering for more accurate time predictions

TheBest6337, 2023-09-04:
**Is your feature request related to a problem? Please describe.**
If you are using a 3d printer with the Klipper firmware you get semi Accurate time predictions, that can be way too short or way to long.

**Which printers will be beneficial to this feature**
Every printer that uses the Klipper firmware

**Describe the solution you'd like**
With the Klipper Estimator post-processing script, it is much closer (About +/-5 seconds) to the actual time. You can use the estimator as a post proccesor, but that comes with the drawback that you only get the estimate after processing (which happens after uploading or saving the gcode) so you don't get the estimate shown in the Slicer and only in your Klipper interface. I would like that this gets implemented so that it's directly in the Orca shown estimate if a printer with Klipper is connected via Mainsail.

**Describe alternatives you've considered**
I've tried to get more accurate estimates by defining the maximum Acceleration, but the estimates also were not near the actual time.

**Additional context**
A link to the Klipper Estimator is here: https://github.com/Annex-Engineering/klipper_estimator

goodwin, 2023-09-11:
As an alternative - compatibility with the existing klipper_estimator as a post-processor will also be good. Currently, Orca is taking estimations from generated gcode before the post-processing script, and not after. Just taking estimation from post-processed output will improve things a lot

samwiseg0, 2023-09-11:
It is possible to tweak motion ability to get very close to klipper estimator. After tweaking orca It is within a few minuets of what klipper estimator would produce.

TheBest6337, 2023-09-12:
> It is possible to tweak "motion ability" to get very close to klipper estimator. After tweaking my orca profile It is within a few minutes of what klipper estimator would produce. 

Fee minutes is not good enough for me. I want that it's almost second accurate (+/- 1 second for me when using the post processor)

discip, 2023-09-21:
Unfortunately as per https://github.com/SoftFever/OrcaSlicer/issues/1612#issuecomment-1666533143 this seems to be not planed. 😞

TheBest6337, 2023-10-12:
Noisyfox might work on an implementation.

closed 2023-10-12 by TheBest6337