OrcaSlicer/OrcaSlicer #2018: Reload from disk on multiple parts

sundewzer, 2023-09-06:
**Is your feature request related to a problem? Please describe.**
When you have multiple parts selected, even if they are the same part there is no option to reload from disk

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
When you have multiple files selected, having a right click option to reload from disk

**Describe alternatives you've considered**
Reload them one at a time or deleting all but one, reload from disk one part, and then pattern again. 

**Additional context**
Prusa slicer has is =P

kayboy-bebop, 2023-11-22:
Related to this request is for 'reload from disk' to remember file location between sessions, to avoid having to navigate to the relevant updated file with every reload.

_tested in Linux Mint_

kinghat, 2024-02-19:
reload from disk isnt working for me with just one object. im using appimage but that shouldnt matter.

crysxd, 2024-02-25:
This would be so handy to have!

kayboy-bebop, 2024-04-29:
> Related to this request is for 'reload from disk' to remember file location between sessions, to avoid having to navigate to the relevant updated file with every reload.
> 
> _tested in Linux Mint_

"Reload from disk" is now persistent between sessions that have been saved, but if loading an autosave after a crash that link seems to be wiped and the file has to be manually picked again.

XoBoR, 2024-05-12:
Having the ability to reload all objects on the current plate or project would be great, having to reload each object individually can be quite a pain...

ColColonCleaner, 2024-07-10:
This would be a very nice feature! Have dozens of objects in a project that use the same base model, so if I change that base model I need to right-click every single one and that takes a while.

ColColonCleaner, 2024-10-09:
Still interested in this, would be great to have.

regularlabs, 2024-12-24:
+1 for being able to reload all selected parts

ChanSmith, 2024-12-26:
Not exactly the same, but there is a 'Reload All' option on plates, added in https://github.com/SoftFever/OrcaSlicer/pull/4903

ColColonCleaner, 2024-12-26:
I'm good with that solution! Thanks!

RsX88, 2025-02-08:
Good enough!

kayboy-bebop, 2025-02-16:
It's great to have a reload all, but if there's one object with a broken file path then it will ask for the file path. Problem being that it doesn't say which object your selecting the filepath for.

My current workaround is to have a dedicated marker mesh (really tall cylinder atm) that I use to replace all the missing meshes, which helps to quickly identify which objects need to individually have their file paths reloaded.

With this little workaround this new feature is a massive timesaver. Thanks!

closed 2025-05-25 by github-actions[bot]