OrcaSlicer/OrcaSlicer #2020: Fan Speed only at 80%

mharmuth, 2023-09-06:
### OrcaSlicer Version

1.6.6

### OS version

macOS

### Printer

Voron 2.4

### How to reproduce

I've got the following filament settings active:
[image]

### Actual results

Fan speed is maxed at 80%
[image]

### Expected results

Fan speed should be at 100% most of the time. Why is it only at 80%

### Project file & Debug log uploads

[3DBenchy.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12535201/3DBenchy.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-09-07:
Seems it's using first filament's fan cooling setting.

mharmuth, 2023-09-07:
You are right. When I remove the first filament, it's working. But I explicitly set the Filament to the second one on the model. The preview correctly shows the color from the second filament, too.

SoftFever, 2023-09-07:
Yes, it's a bug.

SoftFever, 2023-09-07:
Duplicated with [#1829](https://github.com/SoftFever/OrcaSlicer/issues/1829)

closed 2023-09-07 by SoftFever