OrcaSlicer/OrcaSlicer #2021: IDEX printer support

PuszInPusz, 2023-09-06:
Was wondering if you plan on doing a IDEX printer support (specifically some industrial printers like my Craftbot Flow Idex XL) since I switched for everything to OrcaSlicer would be a nice feature to have (having all my printers on one slicer) Would also make OrcaSlicer more professional like IdeaMaler or Simplyfy3D. Since Vorons (Tridex for example) and Ratrigs also started doing Idex stuff it would benefit the community as a whole! I'm pretty sure there are also a lot of BCN3D guys around here who would benefit from a "Slicer made for all tipe of printers".

grengojbo, 2023-09-06:
I'm about to build my second IDEX printer and don't want to use 2 slicers either.

awabom, 2023-11-05:
It is actually already mostly possible to use IDEX printers in OrcaSlicer by treating the 'filament change' g-code as 'tool change' g-code. But, it (like PrusaSlicer) _is_ missing some features like cooldown when idle, turn off when done, preheat, z-hop-on-toolchange, etc.

Servermanden, 2024-01-15:
I'm also interested in IDEX support but for the Sovol SV04
but in general, I think it will bring more people to Orca Slicer

driedger, 2024-01-31:
I am trialing Orca Slicer at the moment, and setup my first printer last night. Im trying to setup my 2nd printer, a TenLog TL-D3 Pro IDEX printer, and am disappointed to learn that not only is it not in the list of device templates, but that Orca Slicer is missing IDEX support completely. Add me to the list of people waiting for IDEX Support please.

Highcooley, 2024-02-03:
+1 for multi extruder, multi material support

MKHTexas, 2024-03-08:
I just got a Vivedino Marathon. Would love to use Orca for this printer as with all my other machines.

allenzha, 2024-03-14:
Please add support for idex Multi-Extruder Multi-Material functionality, thanks!!!

0xmarcelx0, 2024-05-06:
Please add support for idex Multi-Extruder Multi-Material functionality, thanks!!!

riksarchen, 2024-06-06:
Now when Ratrig made idex accessable to the public with the new vc4 I really hope orca will add support. I prefer orca but I'll have to change to prusa otherwise :(

Fisheye3D, 2024-06-10:
> Was wondering if you plan on doing an IDEX printer support (specifically some industrial printers like my Craftbot Flow Idex XL) since I switched for everything to OrcaSlicer would be a nice feature to have (having all my printers on one slicer) Would also make OrcaSlicer more professional like IdeaMaler or Simplyfy3D. Since Vorons (Tridex for example) and Ratrigs also started doing Idex stuff it would benefit the community as a whole! I'm pretty sure there are also a lot of BCN3D guys around here who would benefit from a "Slicer made for all tipe of printers".

Ive been hoping since Orca was released that it would support the craftbot flow idex xl, There craftware and pro slicer are garbage. PS/cura and SS work with the gcode replacement scripting but would love to use Orca.

tlhintoq, 2024-07-06:
2024.07 
Still hoping for IDEX support.  This is not new tech.  I've had these Tenlog printers for years.

tlhintoq, 2024-07-06:
> I am trialing Orca Slicer at the moment, and setup my first printer last night. Im trying to setup my 2nd printer, a TenLog TL-D3 Pro IDEX printer, and am disappointed to learn that not only is it [...]

I've got four of the same machine.  I so very much want to use Orca with them

awabom, 2024-07-06:
Everyone wanting IDEX. Please read my comment earlier:

> It is actually already mostly possible to use IDEX printers in OrcaSlicer by treating the 'filament change' g-code as 'tool change' g-code. But, it (like PrusaSlicer) _is_ missing some features like cooldown when idle, turn off when done, preheat, z-hop-on-toolchange, etc.

Orca is fine to use _right now_ with IDEX. No need to wait.

tlhintoq, 2024-07-06:
red them.  yes *mostly* possible.
That's not the same as truly well supported.

For example:  The part I just sliced will throw in the `T0` and `T1` commands to switch head.  
But didn't correctly identify that the first layer was being done on the `T1` head so the initial heat up is `T0` for no reason.
Then you correctly point out all the missing support like cool down and turning off all the heads when done, et.al.

Yeah - I can go in and edit gCode for every job I slice.  So while its "possible" in OrcaSlicer its not what the ticket describes as "supported".

tlhintoq, 2024-10-05:
Hey bot - don't close me.  This is still a very needed feature. I can't believe that only Cura can do it.  Sad

klack, 2024-10-21:
Wipe tower does not work when using T0 T1 filament changes

AbeFM, 2024-10-27:
I have the HicTop version of the TenLog everyone else has. It's the only reason I keep Cura around.

klack, 2024-10-27:
> I have the HicTop version of the TenLog everyone else has. It's the only reason I keep Cura around.

*Shamless Plug*
For those that do, check out [Luxuri Firmware](https://github.com/klack/LuxuriMarlin) for the 8-bit board.  And join our [Discord](https://discord.com/invite/w8gMzQq8Bp).

We try to stay on top of the latest slicer tricks for IDEX.

Felix14-v2, 2025-01-28:
How this stupid bot works? Still a valid suggestion

Leafy1, 2025-02-12:
Looking for this too. Have my printer all setup with 0.6mm nozzle for fine details and 1.0 for big stuff only to find out I cant configure Orca this way. Really hoping there's some movement on this soon.

Bcjc3, 2025-03-12:
I have a Sovol SV04 that I would love to have supported by Orca Slicer. This is still a valid issue.

GreavesDesignerOne, 2025-03-14:
Another TL-D3 Pro IDEX wrangler here (also a Luxuri Firmware fan). Orca is a great piece of software, but better IDEX support would be greatly appreciated!

Leafy1, 2025-03-14:
Version 2.22 and onward already support this. For some reason version 2.21 for me wasn’t checking for updates even though I don’t remember turning it off, so if that’s the case for you, manually update to 2.22 and you can run separate nozzles.

GreavesDesignerOne, 2025-03-14:
...then moments later I saw [Modbot's video](https://www.youtube.com/watch?v=vQDew7qgdTw) about OrcaSlicer 2.2. Sweet!!
Downloading now.

klack, 2025-03-15:
Somehow I missed the 2.22 release.  This is great news!

Felix14-v2, 2025-03-16:
> Version 2.22 and onward already support this.

Not actually. Orca still doesn't allow you to manage IDEX settings, like splitting the printing area or selecting the print mode — Mirror, Clone (single material modes, double area) or Normal (double material mode, single area). You still have to specify the modes manually in the starting Gcode.

Leafy1, 2025-03-16:
Ok that’s fair. I have separate printer profiles with the correct bed size for mirror and copy modes with the correct code in the print start gcode section for them.

closed 2026-10-10 by SoftFever