OrcaSlicer/OrcaSlicer #2024: Slicer generates unwanted G17 calls and G3 calls with arc fitting disabled

celtare21, 2023-09-06:
### OrcaSlicer Version

1.7.0-dev

### OS version

Windows 11

### Printer

ZeroG Mercury One

### How to reproduce

Slice provided file, check the gcode and random G17 calls are added and also G3 calls even tho "Arc Fitting" is disabled:
[image]

### Actual results

No G3 or G17 calls should be made

### Expected results

G3 and G17 calls are made

### Project file & Debug log uploads

[Mewtwo-Freestanding (2).zip](https://github.com/SoftFever/OrcaSlicer/files/12537096/Mewtwo-Freestanding.2.zip)
[debug_Wed_Sep_06_15_05_23_20396.log.zip](https://github.com/SoftFever/OrcaSlicer/files/12537101/debug_Wed_Sep_06_15_05_23_20396.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-09-06:
@celtare21 
It's the spiral z hop that generate arc moves.
You need to disable that too.

celtare21, 2023-09-06:
@SoftFever just to confirm, this is intended behavior to not have spiral z hop without arc moves?

SoftFever, 2023-09-06:
I'm not sure if I get your question correctly.
The spiral z hop is not linked with arc fitting option. 
They all generate G3 commands but for different purpose.

-Jack

celtare21, 2023-09-06:
Got it, I'll close this ticket then. Thanks for the support

closed 2023-09-06 by celtare21