OrcaSlicer/OrcaSlicer #2025: Plate duplication with objects

jakubfrasunek, 2023-09-06:
Sometimes I need to print two models on top of each other with one extruder and using Z-hop so the result is one final object. I have to align them in slicer as I need them to be positioned, which is fine. But I need to print first the first set of models and then the others set of models on top of the previous one printed that is still on the bed. Would be nice to use multiple plates so I could have one set on one plate and one set on the other. Unable to do it now because of use copy and paste the objects pastes into different position and there is no option to move objects by coordinates like in PS.

So I suggest a feature to be able to duplicate plate which will create new plate with all the objects positioned the same. Just exact duplicate.

closed 2023-12-14 by github-actions[bot]

eoyilmaz, 2024-05-24:
+1 to this feature... aaand I'll give it a try...

eoyilmaz, 2024-05-30:
@SoftFever I'm currently implementing this, I added the extra picker icon to the `Plater` for the `duplicate_plate` functionality, implemented the duplication of the plate along with the objects it contains (still need to work on this part), added the toolbar event etc., but I'm kind of stuck as I get a SEGFAULT whenever I build the code run OrcaSlicer and open the 3D view (GLCANVAS), and I was only able to run it 2 times where I was able to test the duplication of the plate.

Apparently I'm missing something, would you like to review my code if I create a PR?

eoyilmaz, 2024-05-30:
Aah okay found the problem, apparently I missed some lines that needs to be updated from id=6 to id=7 as there is this new icon...

3DQuicky, 2024-11-25:
@eoyilmaz Brilliant! Exactly what I was looking for! This should obviously be default functionality of Orca slicer. Question: How do I go about implementing this solution? Forgive me as I don't have much experience with github.

eoyilmaz, 2024-11-26:
@3DQuicky it is released and avaialable with the latest OrcaSlicer.

But, it is now a menu item under the Edit menu, if you are looking to the videos, they are showing the previous implementation where it was an icon available in the plate actions and in the toolbar.

eoyilmaz, 2024-11-26:
> Is Orca the first slicer to have this epic feature?

Probably, but it should be super easy to back port to Prusa Slicer as it is around 70 lines of code.