OrcaSlicer/OrcaSlicer #2028: Kingroon KLP1 support

Frontier314, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
Support the Kingroon KLP1 CoreXY Klipper Printer

**Which printers will be beneficial to this feature**
Kingroon KLP1 CoreXY Klipper Printer

**Describe the solution you'd like**
To be able to slice and print 3D models for the Kingroon KLP1 printer.

infinisean, 2023-09-14:
Interested in this as well

Frontier314, 2023-09-18:
Kingroon has posted print profiles for Orca for KLP1 and KP3S Pro V2:
https://kingroon.com/blogs/3d-print-101/orcaslicer-using-on-kingroon-kp3s-pro-v2-and-klp1-3d-printer?syclid=ck42f7iusvis73djttc0&utm_campaign=emailmarketing_148241514714&utm_medium=email&utm_source=shopify_email

It would be great if they are included in the next Orca version.

closed 2023-12-26 by github-actions[bot]

snelweg, 2024-08-25:
Would have been nice if they were included.
I learned about Orca from the install of my new printer the KLP1. I tried it and deleted all the Cura stuff I have used for years.
Please include the Kingroon profiles.