OrcaSlicer/OrcaSlicer #2031: Slicer stays on white screen and doesnt open

OrbitalSaint, 2023-09-07:
### OrcaSlicer Version

1.6.6 

### OS version

windows 10

### Additional system information

CPU Intel 11th Gen i7 

### Printer

bambu x1c

### How to reproduce

1. Open slicer 
2. Blank screen

### Actual results

The slicer freezes on white screen. 

### Expected results

The slicer opens 

### Project file & Debug log uploads

theyre blank
[image]
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

lin-ycv, 2023-09-14:
I'm also encountering this problem, to be more exact, for me, this only happens if orca launches on a non-primary display.

closed 2023-12-21 by github-actions[bot]

ultimateshadsform, 2024-02-01:
I have the same issue. on latest version.

[image]

It doesn't freeze but the device tab just doesn't load.

yobaNGE, 2024-10-05:
As mentioned earlier this bug appears when slicer was closed on secondary display, so after you open it again you get a white screen.
For myself i figured a workaround to this, if i catch this bug, disable all secondary screens and start slicer, than turn screens back on.

ultimateshadsform, 2024-10-05:
I only have 1 display.

nicks911e, 2025-08-09:
I just had exactly the same problem in macos, latest os, latest revision of orca slicer, all I did to fix it was drag orca to my primary screen and back again, now all ok