OrcaSlicer/OrcaSlicer #2032: option to hide Bambu Network login window pane

dlarue, 2023-09-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

No problem, just lots of wasted space when a user of the slicer is not on the Bambu network

**Which printers will be beneficial to this feature**
E.g. Voron/Klipper based printer/etc...

All but Bambu printers

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

The left most window pain of the Home screen shows Bambu Network Login/Register and that should be hidden with a config file setting such as: 
        "show_login_window": false
or
        "show_bambu_login_window": false,

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

I've not seen any other way to hide or minimize that window pane.

**Additional context**
Add any other context or screenshots about the feature request here.

johnsonm, 2023-09-07:
It might make sense to actually extend such a setting to also not reach out to e.bambulab.com at all. That might affect the best name to choose for such a feature.

dlarue, 2023-09-07:
> It might make sense to actually extend such a setting to also not reach out to e.bambulab.com at all. That might affect the best name to choose for such a feature.

Does it go to e.bambulab.com regardless of installing the Bambu network driver and/or logging in?  That should not be happening in my book. Thanks for pointing this out

johnsonm, 2023-09-08:
Yeah. I have not installed the Bambu network plugin, never logged into Bambu, don't have a Bambu, so I was surprised that OrcaSlicer was still reaching out to e.bambulab.com. I don't know what for.

szafranski, 2023-09-10:
Hi, I agree. Downloading this Bambu-network-thing should be optional. No point of seeing this, every time after upgrade:
[image]
Thanks!

johnsonm, 2023-12-10:
Not actually stale as of 1.8.1

closed 2024-03-18 by github-actions[bot]

johnsonm, 2024-03-18:
Well, I guess it's now clear that OrcaSlicer is only meant for Bambu. :sob: