OrcaSlicer/OrcaSlicer #2033: Max Volumetric speed bug?

mysticprysm, 2023-09-08:
### OrcaSlicer Version

1.6.6

### OS version

Windows 11

### Additional system information

Ryzen 7 5700x
RTX 3070
32GB RAM

### Printer

BambuLab P1P

### How to reproduce

Change max volumetric speed I think?

### Actual results

So, I'm not sure if it's model specific or what, but when I slice this model, be it with one on the plate, or multiple, it shows this high volumetric speed near the bottom, even though I have it capped. For visual sake, I capped it at 5, and it shows it getting up to 20 still. Unsure if it's a functional bug or visual glitch, but the print does seem to suffer in that area.

### Expected results

The volumetric speed should be capped at whatever you cap it at.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/12555162/log.zip)
[Mushroom Containers.zip](https://github.com/SoftFever/OrcaSlicer/files/12555163/Mushroom.Containers.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ercdvs, 2023-09-28:
Came here to look this up.  I have max volumetric speed set to 24mm^3, but i have, for a .6 nozzle, .3 layer height, and .62 line width have a wall @ 200mm/s

This results in a volumetric limit north of 37mm^3, and therefore fails to print.   The speed should be dropped to enforce this limit

jaimewyant, 2023-09-30:
I'm having a similar issue also.  I've set the max volumetric flow rate to 5 but it hits 13.37 at certain parts of the model.  The image below illustrates this.

[image]

jaimewyant, 2023-09-30:
The bug is fixed in 1.7 for me.  @mysticprysm  may be worth checking out for you too.

ercdvs, 2023-10-01:
1.7 is what i am using, and it is present for me still.  Even though i have max volumetric speed set, its allowing the speed to contribute leading to some areas to outrun my set flow.   I need to go an manually adjust speed for all areas which defeats the purpose

Ninix-source, 2023-10-10:
I'm having the same issues as @ercdvs. Currently running 1.7 too.

top4ek, 2023-10-23:
[image]
Got the same bug on 1.7 while material limit is 8.

hatkyinc2, 2023-11-18:
1.8.0-beta2 same issue, the max volumetric flow doesn't seem to matter if you put speed up, can get to crazy numbers over

hatkyinc2, 2023-11-18:
I found a similar bug in superslicer and it seems correct
https://github.com/supermerill/SuperSlicer/issues/3869
Basically flow multiplier/ratio is applied to max volumetric flow 🤦 
So one should divide the required max volumetric flow by flow ratio to get the desired result

edooman, 2024-02-08:
[image]

Same here on many models, Volumetric test is affected too with abnormal behavior. Orca ver. 2.0.0 dev

closed 2024-05-17 by github-actions[bot]

DearTanker, 2024-07-01:
This error still exists in the latest version of OrcaSlicer V2.1.1 Official Release

jeffalanmcclain, 2024-07-16:
Is it possible that you have "don't slow down outer walls" checked too?  I seem to see this significantly cause external walls to print with faster flow (I think there is not a way to limit it then because too many parameters to obey)?  If I uncheck this option, I can get it to significantly fall into the max volumetric flow rate (time flow ratio) on most areas.  Dramatically lines up.

stefansjs, 2024-08-16:
I just came to check this bug. I'm seeing this as well. I do not have "don't slow down outer walls" checked for this filament. I also calculated the flow rate based on my multiplier of 1.078, which should give me a maximum of 20 mm3/s if I limit to 18.4, but it still shows a maximum of 22.4
[image]

FYI, if the dev team doesn't have resources to fix these types of bugs in a timely manner, then you should consider changing the github actions behavior to not close these issues right away. This is an open source project without (IIUC) any institutional support/money behind it, so the ticket management probably shouldn't be the same as if you had a full-time development team, project managers, etc.

ShantanuNair, 2025-04-01:
@stefansjs @DearTanker @jeffalanmcclain You can see my finings and solutions [here.](https://github.com/SoftFever/OrcaSlicer/issues/7307#issuecomment-2766022845) Only later did I find out @hatkyinc2 already posted here what the issue is. For effective restricting of flow rate, divide your desired max vol speed by your flow ratio.