OrcaSlicer/OrcaSlicer #2034: One wall on top surface does not work reliably when precise wall is enabled

igiannakas, 2023-09-08:
### OrcaSlicer Version

1.7 Dev

### OS version

OSX

### Printer

Bambu X1 carbon

### How to reproduce

Open the attached model

Set wall generator to arachne
Precise wall to checked (on)
One wall threshold (0) - but doesn't make a difference if changed

Observe the issue on the model.
Disable precise wall. Issue on top surface is gone.

### Actual results

With precise wall on
[image]

With precise wall off
[image]

### Expected results

Should precise wall on should work the same way as precise wall off.

### Project file & Debug log uploads

[Issue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12559728/Issue.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jeremytodd1, 2023-09-08:
I think this issue goes a bit deeper than just the Precise Wall feature, unfortunately.

I've been noticing artifacts on top layers with ever since the "only one wall on top surfaces" feature was updated to work on the Arachne slicing engine. This will still happen if the Price Walls features is disabled or enabled.

**The only way that the top layer will not have these artifacts is to fully disable the "One wall on top layers" feature, which definitely isn't ideal.**

Here are a couple pictures on a simple model to show the issue. This will show that the setting isn't just simply the Precise Wall feature. 

**Only one wall on top surfaces:** Disabled
**Precise Walls:** Enabled
**Arachne:** Enabled
These settings will always product top layers that have no artifacting.
[image]

**Only one wall on top surfaces:** Enabled
**Precise Walls:** Enabled
**Arachne:** Enabled
These settings will generate some artifacting on top layers.
[image]

**Only one wall on top surfaces:** Enabled
**Precise Walls:** Disabled
**Arachne:** Enabled
These settings will generate some artifacting on top layers.
[image]

**Only one wall on top surfaces:** Enabled
**Precise Walls:** Enabled
**Arachne:** Disabled
These settings will generate some artifacting on top layers.
[image]

bpkenn, 2023-09-18:
I am seeing similar behavior with "only one wall on top surfaces"
[image]

With the feature turned off:
[image]

samwiseg0, 2023-09-18:
@jeremytodd1 @bpkenn Have you played with the one wall threshold? You should be able to lower it and get your desired results. I do not think your issue is related when "Precise wall" is disabled. This is a similar issue that was always present with SuperSlicer and the same feature. 

[image]

As for OPs orginal problem changing the threshold does not seem to help so it maybe a bug when both features are on.

jeremytodd1, 2023-09-18:
@samwiseg0 

I have, yes. It's not able to get it quite right no matter what the threshold is set to.

My default one wall threshold is set to 300%. 
I've played around with it quite a bit on this object that I referenced in my last post.
156% is the lowest I can go before it'll start showing multiple perimeters on the top surface on some parts of the object.
157% is when single perims does cover the entire object, which is good, but I am still seeing some artifacts on the top layer, unfortunately.

samwiseg0, 2023-09-18:
> @samwiseg0
>
> I have, yes. It's not able to get it quite right no matter what the threshold is set to.
>
> My default one wall threshold is set to 300%. I've played around with it quite a bit on this [...]

Interesting using your model I had no issues at 50%
[image]
[jeremytodd1.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/12654148/jeremytodd1.3mf.zip)

jeremytodd1, 2023-09-19:
@samwiseg0 (I'm not sure if you get notified of my reply if I don't tag you lol. Maybe this tagging isn't necessary)

Ah, I had price walls on during my testing.

50% One Wall Threshold
Price walls Enabled
[image]

50% One Wall Threshold
Price walls Disabled
[image]

Can you check to see what yours done if you enable precise walls?

samwiseg0, 2023-09-19:
I do get notified since I am participating in the issue. Tagging is not necessary unless you want to speak directly to someone or want to ping someone not participating :)

Yep "precise wall" does cause some artifacts on the top layer using your model. I was just pointing out that I believe the issue as OP described is only when both are on. 

Your and @bpkenn issue when "precise wall" is **_off_** is related to "one wall threshold" needing to be tweaked to get the desired results on the top layer. 

I am not sure if this is simply an incompatibility of both features being on at the same time or a bug that can be fixed. SoftFever describes what "precise wall" does [here](https://github.com/SoftFever/OrcaSlicer/discussions/831#discussioncomment-6274332). I personally leave it off as I am not sure if it would have a significant impact on making the parts more dimensionally accurate.

bpkenn, 2023-09-19:
Pulling back from the default of 300% fixed my issue (it varies based on the model).  Was the feature changed recently?  I had never needed to adjust this previously, and did not experience the behavior until just recently.

samwiseg0, 2023-09-19:
> Pulling back from the default of 300% fixed my issue (it varies based on the model). Was the feature changed recently? I had never needed to adjust this previously, and did not experience the behavior until just recently.

That would depend on which version you were running before. The last improvement to the feature was this https://github.com/SoftFever/OrcaSlicer/issues/1680 Which was released in V1.6.4 Beta 2

bpkenn, 2023-09-19:
> That would depend on which version you were running before. The last improvement to the feature was this #1680 Which was released in V1.6.4 Beta 2

Yup that probably explains it, I was on 1.6.3 and then went straight to 1.7

igiannakas, 2023-12-21:
Closing as this has been mostly fixed in the latest nightly.

igiannakas, 2023-12-21:
Following the fix in the precise wall mode earlier this month.

closed 2023-12-21 by igiannakas