OrcaSlicer/OrcaSlicer #2041: Show Export G-code button directly

Brugman, 2023-09-08:
When using a non-networked printer, after every slice, we are confronted with a greyed out Print button.
Instead of 1 click to get the g-code, it now takes 3, of which the first is on the relatively tiny dropdown button.

I'd like to have Orca recognize that we do not have a networked printer, and that we just sliced something, so because the most likely next step is storing exporting the g-code manually, that it selects this option automatically.

[image]

foreachthing, 2023-09-08:
I do like how Prusa does it: when a SD-Card is in, it will "automatically" save to it.

niklasb22, 2023-09-26:
I just tried to recreate it. For me, if no network configuration is specified for a printer, the G-Code export is displayed by default.

See also "src\slic3r\GUI\MainFrame.cpp" (L. 1075)

Brugman, 2023-09-26:
Yeah. I think 1-2 days after posting this issue it magically started working as I requested. Could have been implemented quietly.

SoftFever, 2023-11-19:
Yeah, this had been implemented and I forgot to update it here...

Close this ticket as it is fixed.

closed 2023-11-19 by SoftFever