OrcaSlicer/OrcaSlicer #2042: Option to disable Device tab

Brugman, 2023-09-08:
With a non-networked printer, we're never going to use the Device tab.

I appreciate the instructions the tab shows on how to connect your networked printer, but if you don't have one, you don't need those either. So after the user recognizes that this tab is not useful to them, I'd like to hide it from the UI with a checkbox in the preferences window. Having the tab enabled by default is fine.

Brandon168, 2023-09-14:
Another reason: Webcam streaming is happening in the background. With a few instances of OrcaSlicer open I see my bandwidth and cpu taking a hit when using the default mjpeg setting in crowsnest.

closed 2023-12-21 by github-actions[bot]

dderg, 2023-12-26:
I'd like to see this too. Noticed that battery life of my laptop decreased when I have the slicer open.

vertigo235, 2024-01-06:
I noticed my PC runs super slow as I'm trying to migrate to Orca Slicer, and I feel like this might be the culpret.   This was closed as "completed" but is it?